ArduPilot/ardupilot #5131: Copter: too fast taking off in Guided_NoGPS

korigod, 2016-11-03:
#### Issue details
We have an issue using Guided_NoGPS mode. When we try to take off sending SET_ATTITUDE_TARGET with thrust > 0.5, copter takes off very fast and stop ascending only at 3-4 meters even if thrust is set to 0.5 or lower immediately after lift off.
It happens even when thrust only slightly exceeds 0.5 (e.g. thrust = 0.501).

#### Version
APM:Copter 3.4.1.

#### Platform
[   ] All
[   ] AntennaTracker
[X] Copter
[   ] Plane
[   ] Rover

#### Airframe type
Octocopter.

#### Hardware type
Pixhawk.

OXINARF, 2016-11-03:
This should typically be posted first in the forum since this issues list is for confirmed bugs and feature requests.
But since you opened it please provide a log.

OXINARF, 2016-12-17:
No answer for one month and a half, closing.

closed 2016-12-17 by OXINARF