ArduPilot/ardupilot #5138: GPS_INPUT in ReleaseNote

icyore, 2016-11-04:
Copter 3.4.0-rc2 08-Aug-2016 version ReleaseNote:
10)  d) support for GPS_INPUT mavlink message to allow supplying vehicle position data from other sources

However, when I download the firemare from the missionplanner, the 3.4.0 rc7 and 3.4.1 don't respond to message I sent. I compile the firemare 3.5 by myself, it works. The problem take me a week to check the code of mavros and the parameters. Is it problem of missionplanner? 

Copter 3.4.0-rc2 - Copter 3.4.1

quadrotor 
pixhawk

magicrub, 2016-11-04:
It is possible that it's in master and not on the copter 3.4.x branch. Does anyone know if that's the case? I can check later today if no one responds

OXINARF, 2016-11-04:
GPS_INPUT is certainly supported in 3.4. If it is working in master and not in the 3.4 releases then we might have a bug.

@rmackay9 can you confirm this?

rmackay9, 2016-11-04:
@icyore it should work because I used it for this recent flight with the Pozyx sensor.  https://www.youtube.com/watch?v=MKWlPsBmAyo

I suspect what you're missing is you need to set the GPS_TYPE parameter to 14.  these are the full parameter changes that I made for my flight.
GPS_TYPE,14    <-- sets driver to MAV_GPS driver
SERIAL2_BAUD,115   <-- set telemetry2 baud rate to 115200
SERIAL2_PROTOCOL,1   <-- set telemetry2 protocol to Mavlink
BRD_SER2_RTSCTS,0   <-- turn off telemetry2's flow control (probably not necessary)
EK2_GPS_TYPE,2  <-- sets EKF's GPS type to only use 2D position because my pozyx doesn't supply 3d position or velocity.

icyore, 2016-11-05:
GPS_TYPE,14
SERIAL2_BAUD,57 or 115
SERIAL2_PROTOCOL,1 
BRD_SER2_RTSCTS,0 
v3.4.1(04c76843) downloaded by mission planner , the LED don't turn GREEN. 
v3.5-dev(f66be99f) compiled by myself, the LED turn GREEN.

rmackay9, 2016-11-07:
Ok, I'll include this commit in AC3.4.2-rc1 so maybe you can test the next beta that comes out in a day or two to see if that resolves the issue (it should).
https://github.com/ArduPilot/ardupilot/commit/9290ee65cc96871ef110282fe54dbb8a1be070fd

rmackay9, 2016-11-07:
@icyore, the new beta (AC3.4.2-rc2) is out now so if you could give it a try that'd be great, thanks!

icyore, 2016-11-08:
I have tested on V3.4.2-rc2(179a7f57), It is OK.

rmackay9, 2016-11-08:
Great!  Thanks for testing.

closed 2016-11-08 by rmackay9