ArduPilot/ardupilot #5146: Allow a combination of alt sensors for precision landing

expntly, 2016-11-05:
@jschall @rmackay9 

#### Issue details
Precision landing currently relies 100% on a rangefinder.  No rangefinder on the vehicle means no precision landing.  Similarly, when a rangefinder is installed, precision landing won't kick in until the rangefinder provides sufficiently good data.  What about relying on all altitude calculating sensors for precland instead of just the rangefinder?

Here's a graph that shows what the problem is: https://dl.dropboxusercontent.com/u/67653/share/precland.png.  Essentially, a (cheap) sonar will only provide good data pretty low above the ground, for mine from 4m down to 0.2m.  However my camera sees the target way up at 25m+.  So when descending, from 25m to 4m, the precision landing code isn't used because my sonar hasn't yet kicked in in providing any or good data.  Even if my sonar worked as advertised up to 7m, the problem would persist.  What I end up seeing is the vehicle completely missing the target and not being repositioned when above "good sonar range" even though the camera has a solid lock on the target.

Would it be possible to use other altitude measuring sensors until the rangefinder kicks in?  Looking at the graph and POS.RelAlt shows that the FC does a pretty decent job at estimating actual altitude with all this noise.  Would it be possible to use this altitude instead of raw rangefinder data?

#### Version
commit 0bcbf726acc86cfd6eae4929cdd65ef31c535818

#### Platform
[  ] All
[  ] AntennaTracker
[  ] Copter
[X] Plane
[  ] Rover

#### Airframe type
quadplane

#### Hardware type
pixhawk

#### Logs
I can produce clean logs if needed.

OXINARF, 2016-11-05:
> What about relying on all altitude calculating sensors for precland instead of just the rangefinder?

What other sensor gives you an altitude above ground?

expntly, 2016-11-05:
As the graph shows, the FC makes a good job at figuring out the altitude above ground without the help of the rangefinder.  Rough guess is sub-meter accuracy.  The good news is, you only really need the rangefinder close (a few meters?) to the ground, and that's when it does a good job.

In other words, here's what I'm suggesting, picturing a descent:
 - for the range [max_camera_target_acquired_alt ; max_rangefinder_dist]  : use altitude calculated by the FC (some kind of POS.RelAlt)
 - for the range [max_rangefinder_dist ; 0] : use altitude calculated by the rangefinder

Obviously it doesn't have to be that binary and could probably use some linear smoothing, but basically the idea is to use all sensors and give more weight to the rangefinder only as we get closer to the ground.  Not to exclusively rely on a rangefinder.

OXINARF, 2016-11-05:
Without a rangefinder there is no way to measure altitude above ground (well, you can have SRTM data but that's too inaccurate for this). What the FC is calculating is the altitude above home, not above ground.

https://gitter.im/ArduPilot/ardupilot?at=581d437c45c9e3eb43fd13c6 - there you have a good explanation from Randy why the rangefinder is needed.

This will probably get closed, but I want you to understand why this isn't feasible. If you don't, please reply here or in Gitter so that we can try to explain it better.

magicrub, 2016-11-05:
This is for a quad plane which is plane domain, not copter. Things don't
work exactly the same on landing. In fact, the landing code is completely
different.

I'm still not sure what you're asking for. If you need longer range, don't
use sonar. It is a short range sensor. Upgrade to lidar.

magicrub, 2016-11-05:
What I'm confused about is that your request describes how it is already
currently implemented. Please compare how it works now to how you'd like it
to work. For me, I see no difference.

OXINARF, 2016-11-05:
@magicrub He wants precision landing to work without a rangefinder (or above the maximum height of the rangefinder).

magicrub, 2016-11-05:
A camera does not tell you an altitude, only a velocity via optical flow. You need two camera and lots of cpu power to get an altitude and that altitude accuracy did add your altitude goes up. What you need is lidar and a properly tuned aircraft and a properly planned mission. I'm closing this. Please post on the forums discuss.ardupilot.org about how to tune your aircraft and plan a good landing mission and supply your param file with an example mission. You can also tag me so I can take a look.

expntly, 2016-11-05:
@magicrub Appreciate you taking a look, allow me to reply and reopen, there's already context in this Git issue and this can be used to discuss whether a code change can/should be implemented.  I'm willing to spend time on it.  I've already spent the last few weeks on precland anyways.

What I'm trying to figure out is whether there's a way to rely on non-rangefinder altitude signals **for the upper portion** of the precision landing.  (For the lower portion, i.e. a few meters above, I have no doubts a rangefinder is needed.)  I'm inclined to think that it should be practical as it was mentioned the FC has a rough idea how high AGL it is.  @peterbarker on gitter mentioned "It measures pressures and temperatures before takeoff. It also uses the SRTM database to try to guess terrain elevation".  Are there cases I'm missing when the FC will be way off on its approximation of AGL alt?  Outside of apocalyptic scenarios that is :-)

I'll dig into the code but if you can point me at the place that deals with returning AGL altitude, that'd be helpful to start and I can run some tests, thanks!

magicrub, 2016-11-06:
Sounds like what you want is accurate altitude at highish altitude. Since the terrain below you is changing, then the alt is generally relative. It's relative to your takeoff point, nothing else. Srtm data will tell you the relaxation of the ground, to a few meters accuracy, but then you must friend on a bars which has drifted and your GPS with huge altitude error.

I'm open to suggestions but your camera idea won't be any more accurate than baro/GPS and requires huge resources to accomplish. 

Get a lidar.

rmackay9, 2016-11-06:
@jschall has talked about adding an estimate of height based upon the movement of the target.  He seems to think it's possible.

expntly, 2016-11-09:
Sounds great!

Here's another idea that requires no API change.  When the camera sees the target and the vehicle is too high for the sonar, just disregard height completely and use the same mavlink LANDING_TARGET's angle_x and angle_y to move the vehicle above the target in a pretty conservative way.  It obviously won't work as well as when the height is available, but at least it will move the vehicle in the horizontal plane so it will be in a better position for when the sonar starts reading.  A slight improvement would be to use SRTM's rough height estimates.

expntly, 2016-11-21:
One more idea.  The LANDING_TARGET message takes a `distance` param which is ignored today.  This could be used when the rangefinder doesn't provide data.

fnoop, 2017-03-30:
@expntly This is implemented now for copter:
https://github.com/ArduPilot/ardupilot/issues/5636

rmackay9, 2017-03-30:
@fnoop, you know.. it might work without lidar if we change this line:
https://github.com/ArduPilot/ardupilot/blob/master/ArduCopter/precision_landing.cpp#L27
From:

- copter.precland.update(height_above_ground_cm, rangefinder_alt_ok());

To this:

- copter.precland.update(height_above_ground_cm, true);

OXINARF, 2017-03-30:
@rmackay9 I wouldn't do that. I'm not sure why it tries to use the terrain altitude now since that is then rejected inside the PrecLand library.

It seems very dangerous to me to use an inaccurate AGL to land, specially while using that AGL to calculate an horizontal move - for terrain following it might be OK to use SRTM, but not if you are flying low.
Maybe we could use height above home if we are close though.

amilcarlucas, 2017-09-25:
For applications with RTK GPS and when landing at home, GPS height might be a good source as well

IamPete1, 2021-08-30:
@rishabsingh3003 can we do this now?

rishabsingh3003, 2021-08-31:
@IamPete1 if you are using LANDING_TARGET mavlink message with the DISTANCE field filled in then yes. You don't need a downward-facing rangefinder to prec land (although it is recommended because some features won't work).
However, it will be needed for IRLock

IamPete1, 2021-09-11:
Please re-open if this is still a issue.

closed 2021-09-11 by IamPete1