ArduPilot/ardupilot #5148: LogAnalyzer: TestDupeData has variable-type related bug

Georacer, 2016-11-05:
#### Issue details
TestDupeData test from LogAnalyzer uses the `logdata.channels["ATT"]["Pitch"].listData` list to store the pitch data of the aircraft. However, for some reason the stored data are integers, instead of floats for pitch, eg, the first twenty rows of my log are:
```
[(441, 0), (569, 0), (698, 0), (825, 0), (918, 1), (964, 1), (985, 1), (1007, 1), (1035, 1), (1056, 1), (1085, 1), (1107, 1), (1128, 1), (1156, 2), (1177, 2), (1202, 2), (1223, 2), (1244, 2), (1272, 2), (1293, 2)]
```
As a reminder, the first element is the line of the log file where the ATT/Pitch data is recorded and the second is the pitch value.

As a result, when checking for duplicate data, the integer precision makes the test fail more often than needed.

For example, the test returns the  message:
```
Duplicate data chunks found in log (315109 and 318033)
```
when in fact the contents of the corresponding rows are:
```
ATT, 1478137058, -4.48, -9.98, 1.37, 1.29, 0.00, 7.72, 0.03, 0.02
```
and
```
ATT, 1475677295, -7.42, -6.30, 1.67, 1.80, 0.00, 17.14, 0.03, 0.05
```

#### Version
N/A

#### Platform
[  ] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover

#### Airframe type
N/A

#### Hardware type
N/A

#### Logs
_Please provide a link to any relevant logs that show the issue_

peterbarker, 2016-11-07:
@Georacer Where did that log file come from?

If you have a look at the format line at the top, you will probably find that the type designator for the Pitch column is "c", indicating that the number in that column can be stored as a C -integers.

Sadly, your data lines there are obviously not C integers.  This probably comes down to the log dumping code in ArduPilot, which (incorrectly IMO) performs a conversion on the column value on the way out: https://github.com/ArduPilot/ardupilot/blob/master/libraries/DataFlash/LogFile.cpp#L438  This puts it at odds with the FMT message which it has previously emitted.

I'm really not sure how to fix this.  I don't think we can stop changing values on the way out -  the GCS out there can't be scaling them ATM (just relying on the fact they are pre-scaled) when doing their graphing.

Perhaps I could munge the format-string line on the way out to be a float rather than int*100?

Alternatively, we could kill the log-dump feature with fire and ask people to use download the binary logs!

(note that TestDupeData is probably of no use to you; I believe it was really just looking for an issue you don't get on SD cards).

Georacer, 2016-11-07:
@peterbarker 
Here is a Dropbox link to the DF log file (it was a bit large to attach) https://www.dropbox.com/s/6uqyx8fvvamfu2m/68.BIN?dl=0

Here is my workflow:
I have used Mission Planner to convert from .BIN to a human-readable .CSV file (https://www.dropbox.com/s/0eg4dsvrbchex40/68.log?dl=0).
This is where these lines I posted are from.

So let me re-state what you said to see if I understand it correctly:
1. The "ATT/Pitch" field (among many others) is declared to be a 'c'-type, which corresponds to an integer multipled by 100, as results from [here](https://github.com/ArduPilot/ardupilot/blob/master/libraries/DataFlash/LogStructure.h#L719)
`FMT, 167, 27, ATT, QccccCCCC, TimeUS,DesRoll,Roll,DesPitch,Pitch,DesYaw,Yaw,ErrRP,ErrYaw`
2. However, when performing the conversion to .CSV, the library (and by extension Mission Planner) divides it by 100 and casts it into a float.
3. I am using these float numbers to perform my processing.

However, from this point forward, I fail to follow your logic. If LogAnalyzer code uses the raw .BIN file to test for duplicate data, it should compare int16*100 numbers for equity which is fine.
On the other hand, casting these numbers into float/100 and comparing them for equity should also be fine (down to a reasonable epsilon).

But what I notice is that LA is parsing on truncated int16/100 data which obviously lacks the precision to perform the comparison, e.g. 167/100 is turned into a 1, instead of a 1.67.

Have I got something wrong here?

---

Regardless of the above, I am currently going thorough the LA codebase to get a feeling for what it does, because I am writing a MATLAB port right now.

I don't think we should nuke the DF protocol, just post it somewhere in BIG letters and fix wrong iterpretations, such as the `print` function you linked at. The GCSs should have access to a stable format definition and work accordingly.

That is, assuming there was a reason to that weird `c` definition (int*100), which I guess was because of GPS precision and the reluctance to use a float when a scaled int can do the job. If it was arbitrary, then we are shooting ourselves on the foot here.

Regardless of the above, I will start using the raw .BIN file from now on. Thanks for the info!

peterbarker, 2016-11-07:
On Mon, 7 Nov 2016, George Zogopoulos Papaliakos wrote:

> However, from this point forward, I fail to follow your logic. If
> LogAnalyzer code uses the raw .BIN file to test for duplicate data, it
> should compare int16*100 numbers for equity which is fine.
> On [...]

No, that's quite right.  LA isn't doing anything wrong; it has been told 
to interpret a field as an integer, and it is.  The field value itself is 
malformed!

> Regardless of the above, I am currently going thorough the LA codebase to
> get a feeling for what it does, because I am writing a MATLAB port right
> now.

Cool.  You might also look at dronekit-la.

> I don't think we should nuke the DF protocol, just post it somewhere in BIG
> letters and fix wrong iterpretations, such as the print function you linked
> at. The GCSs should have access to a stable format definition and work
> accordingly.

The FMT message (and the general structure of the DF log) are pretty-much 
set in stone.  These text-bsaed log dumps are an anachronism.  I can't 
really fault MissionPlanner for emulating what ArduPilot does itself!

> That is, assuming there was a reason to that weird c definition (int*100),
> which I guess was because of GPS precision and the reluctance to use a float

Yep, I believe that's why.

peterbarker, 2017-12-19:
I don't think there's more to do on this issue.

closed 2017-12-19 by peterbarker