ArduPilot/ardupilot #5150: SRXL support for Multiplex receivers

RomanSeb, 2016-11-05:
#### Issue details
At the moment SRXL for Multiplex receivers is not supported via DSM port. As fas as I know, all SRXL variants use the same frame structure, Baud-Rate and UART configuration.  It shall be possible to implement all variants without interference.

My request is to implement the Multiplex SRXL variants alongside the recently implemented Spektrum SRXL variant. 

Multiplex SRXL Documentation can be found here:
http://www.multiplex-rc.de/Downloads/Multiplex/Schnittstellenbeschreibungen/srxl-multiplex-v2.pdf
The 12-channel Multiplex SRXL variant "v1" is identical to JR PROPO X-BUS Mode B. This means JR receivers are supported over DSM port as well, if Multiplex SRXL is supported by PX4 Firmware. 

This is the SRXL frame structure is shared among all SRXL variants
 - Header byte
 - Payload
 - 16Bit CRC over Payload and header byte 

SRXL variants can be distinguished based on their header byte:
 - 0xA1: Multiplex SRXL V1 "12 channel" and X-Bus Mode B
 - 0xA2: Multiplex SRXL V2 "16 channel"
 - 0xA5: Spektrum 

SRXL variants can also be distinguished based on the number of payload bytes
 - 24: Multiplex SRXL V1 "12 channel" and X-Bus Mode B
 - 32: Multiplex SRXL V2 "16 channel"
 - 16: Spektrum 

#### Version
All versions

#### Platform
[  ] All
[  ] AntennaTracker
[ x ] Copter
[ x] Plane
[ x] Rover

#### Airframe type
Quad

#### Hardware type
Pixhawk

#### Logs
no logs yet

RomanSeb, 2016-11-05:
Hello,
I added Multiplex SRXLv1 and SRXLv2 support and pulled a request for PX4Firmware module
 - #70 SRXL support added for Multiplex SRXL v1-12channel and v2-16channel 

BR

Code-Tester, 2016-11-05:
# Precision Landing does not work with RTL
When you have a look at the code file "control_rtl.cpp" you can see the class "Copter::rtl_land_run()" with the special functions from the normal "land-mode" inside:

land_run_horizontal_control();
land_run_vertical_control():

And in the class "Copter::land_run_horizontal_control()" i see the important lines

#if PRECISION_LANDING == ENABLED
    bool doing_precision_landing = !ap.land_repo_active && precland.target_acquired();
    // run precision landing
    if (doing_precision_landing) {
        Vector2f target_pos, target_vel_rel;
        if (!precland.get_target_position_cm(target_pos)) {
            target_pos.x = inertial_nav.get_position().x;
            target_pos.y = inertial_nav.get_position().y;
 and so on...

So my question is, why does it not work and why ?
Have I overlooked something ?

OXINARF, 2016-11-24:
This has been fixed by @RomanSeb himself (with an awesome PR!) and was just merged by @tridge.

closed 2016-11-24 by OXINARF

pedrodparkes, 2018-03-04:
hey, I use JR x-bus receiver in mode B.

It works well with ArduPlane but not with PX4 firmware.