ArduPilot/ardupilot #5152: Precision Landing does not work with RTL

Code-Tester, 2016-11-05:
#### Issue details

When you have a look at the code file "control_rtl.cpp" you can see the class "Copter::rtl_land_run()" with the special functions from the normal "land-mode" inside:

land_run_horizontal_control();
land_run_vertical_control():

And in the class "Copter::land_run_horizontal_control()" i see the important lines

#if PRECISION_LANDING == ENABLED
bool doing_precision_landing = !ap.land_repo_active && precland.target_acquired();
// run precision landing
if (doing_precision_landing) {
Vector2f target_pos, target_vel_rel;
if (!precland.get_target_position_cm(target_pos)) {
target_pos.x = inertial_nav.get_position().x;
target_pos.y = inertial_nav.get_position().y;
and so on...

So my question is, why does it not work and why ?
Have I overlooked something ?

#### Version
ArduCopter V3.4-rc7 until master

#### Platform
[  ] All
[  ] AntennaTracker
[ x ] Copter
[  ] Plane
[  ] Rover

#### Airframe type
Quad-Copter

#### Hardware type
Pixhawk V1

#### Logs
No Logs at the moment, only visual test

lvale, 2016-11-05:
Do you have a Lidar or similar?

Code-Tester, 2016-11-05:
Yes i have a Lidar. SF11/C from Lightware over Serial-Connection. Also the Precision Landing works already in the Land-Mode. Did you see a reason why it can't work together with RTL?

OXINARF, 2016-11-06:
Looking at code it looks like it works. As you said it uses the same code in Land and RTL.

This issues list is for confirmed bugs or feature requests but this is a support question. Those should be posted at the forum (http://discuss.ardupilot.org) together with a log so that it can be checked.

closed 2016-11-06 by OXINARF