ArduPilot/ardupilot #5153: Rangefinder: Ground Clearance will be not substract

Code-Tester, 2016-11-06:
#### Issue details
We defined a "Ground Clearance" in the parameters for the Rangefinder, but doesn't use this paramter in the code to calculate the difference for the real altitude.

Normally every Copter has a landing gear. And the Rangefinder is mounted higher than the bottom of the landing gear. And for that case we must substract the "ground_clearance_cm()" from the "rangefinder_state.alt_cm"

So my Idea:

in the ArduCopter/sensors.cpp we should change the follow line number 44:

from "int16_t temp_alt = rangefinder.distance_cm();"
to "int16_t temp_alt = rangefinder.distance_cm()-rangefinder.ground_clearance_cm();"

and also we should prevent negative values in the follow line number 51:
from  "rangefinder_state.alt_cm = temp_alt;"
to "rangefinder_state.alt_cm = MAX(temp_alt, 0.0f):"

Also we must add a ground clearance in cm for the IR-Lock sensor in combination with precision landing.

**The current code assumes the same sensor level (LIDAR and IR-LOCK) and does not consider the ground clearance of the landing gear**

I think this is a big bug.

Hey guys what did you think about these solutions. Is it possible to fix this?

#### Version
ArduCopter V3.4-rc7

#### Platform
[  ] All
[  ] AntennaTracker
[ x ] Copter
[  ] Plane
[  ] Rover

#### Airframe type
Quad-Copter

#### Hardware type
Pixhawk V1

#### Logs
No Logs

rmackay9, 2016-11-07:
Thanks for the report.

We need to be consider a few things:
- is it really important that the distance reported has the ground clearance subtracted?  It's not completely clear (to me at least) why reporting the distance to the ground from the bottom of the landing gear is better than reporting the distance to the ground from the body of the vehicle (which is where the lidar/sonar is likely connected).
- the precision landing probably actually wants the distance from the lens to the target, not the distance from the bottom of the landing gear although I'm not totally sure.
- should the ground clearance be subtracted before or after applying the vehicle's lean angle correction?

magicrub, 2016-11-07:
Subtraction is for sure needed for plane but you bring up good points for
copter.

R-Lefebvre, 2016-11-13:
Probably makes sense to have two data available. Ground clearance subtracted height, and then the raw sensor height (to be used for PrecLand).  

On large aircraft, the sensor height can be quite high.  In future, if we have really good distance data to the actual landing gear, could be used for things like positively not-auto-disarming while not actually landed.