ArduPilot/ardupilot #5160: Copter: wiki updates for AC3.5

rmackay9, 2016-11-07:
The list of wiki page updates that should be completed before AC3.5 is released:
- [ ] flight time recorder
- [x] servo gripper: http://ardupilot.org/copter/docs/common-gripper-servo.html
- [ ] how to specifying sensor offsets for EKF (GPS, compass)

Updates for AC3.4 can be seen here: https://github.com/ArduPilot/ardupilot/issues/4674

magicrub, 2016-11-07:
Leddarone lidar

magicrub, 2016-11-07:
Btw, wouldn't this be best done as an issue in the wiki issue trackerr?

rmackay9, 2016-11-07:
Leddar one is actually already done for AC3.4: http://ardupilot.org/copter/docs/common-leddar-one-lidar.html

I like having this item here because it's easier to include (I think) with our code issues which are the main drivers for the release.

magicrub, 2016-11-07:
I'll update the release version on that page to v3.5

OXINARF, 2016-11-07:
@magicrub LeddarOne will be in 3.4.2

magicrub, 2016-11-07:
Yeah, that's what I meant 😀

magicrub, 2016-11-08:
wiki LeddarOne updated to v3.4.2: https://github.com/ArduPilot/ardupilot_wiki/commit/04a799c96352b1736f34bc99f77e701dfb9e8236?w=1

patrickpoirier51, 2017-01-10:
Suggestion:
Update the optical flow page by adding the Garmin Lidar Lite v3  (I2C mode)
Remove PulseLight because this product no longer exist.,
Keep a reference for pulselight  name  because Lidar Lite is the subsitution 
Option: Renaming pulselight  to lidarlite on the code ?

RQRobert, 2017-07-15:
Kindly update

"dodeca-hexa esc wiring and propeller direction in 

Connect ESCs and Motors Wiki Page

tonysmartantzoulatos, 2019-04-04:
Is there a way to set flight time recorder to produce a warning message on MP if a certain flight time has been reached in order to maintain periodic checks on vehicles?

rmackay9, 2019-04-05:
@tonysmartantzoulatos, it's possible to add that but it's not a feature we have at the moment.  Feel free to raise a new enhancement request although my feeling is that this should be a feature of a ground based drone maintenance application.. i.e. a fancy GCS or server based system that keeps an eye on all the drones a person or company owns and automatically analyses logs looking for problems.  I haven't seen too many of these types of programs but I suspect they exist.

rmackay9, 2020-02-06:
I think I'll close this now because it's quite old.  We are maintaining wiki to-do items on the wiki issues list now: https://github.com/ArduPilot/ardupilot_wiki/issues

closed 2020-02-06 by rmackay9