ArduPilot/ardupilot #5161: Possible memory leak from usage of strdup withou free

khancyr, 2016-11-07:
#### Issue details
In some place in the code strdup is use but according to strdup doc, it use malloc, so memory allocated must be free explicitly. 

Some example in code : 
https://github.com/ArduPilot/ardupilot/blob/c808ee2f497e7a7a6d570632a3224e655b377323/libraries/AP_HAL_PX4/HAL_PX4_Class.cpp#L274

https://github.com/ArduPilot/ardupilot/blob/c808ee2f497e7a7a6d570632a3224e655b377323/libraries/AP_HAL_Linux/UARTDriver.cpp#L140

https://github.com/ArduPilot/ardupilot/blob/c808ee2f497e7a7a6d570632a3224e655b377323/libraries/AP_HAL_SITL/SITL_cmdline.cpp#L229

https://github.com/ArduPilot/ardupilot/blob/c808ee2f497e7a7a6d570632a3224e655b377323/libraries/AP_HAL_Linux/PWM_Sysfs.cpp#L231

Other place see : https://github.com/ArduPilot/ardupilot/search?utf8=%E2%9C%93&q=strdup

Am I missing something or it really miss free() call ?

#### Version

Present in https://github.com/ArduPilot/ardupilot/commit/ae8aea482f9b941f5faff1c863106042b0b68653

#### Platform
[X] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover

tridge, 2016-11-07:
These examples are all for strings that are used for the rest of the execution of ArduPilot, so we can't free them as we still need them.
If you find examples where we don't need the string then you could let us know

khancyr, 2016-11-07:
ok thanks for the answer !

lucasdemarchi, 2016-11-07:
On exit everything is freed... but if you do this, it's harder to use tools like valgrind to spot allocation problems since you now have gazillion "allocated and not freed" problems in the report.

Freeing on class destructor is an option.

lucasdemarchi, 2016-11-08:
> https://github.com/ArduPilot/ardupilot/blob/c808ee2f497e7a7a6d570632a3224e655b377323/libraries/AP_HAL_Linux/PWM_Sysfs.cpp#L231

This is fixed in the PR referenced above.

> https://github.com/ArduPilot/ardupilot/blob/c808ee2f497e7a7a6d570632a3224e655b377323/libraries/AP_HAL_Linux/UARTDriver.cpp#L140

The string is being freed.

> https://github.com/ArduPilot/ardupilot/blob/c808ee2f497e7a7a6d570632a3224e655b377323/libraries/AP_HAL_SITL/SITL_cmdline.cpp#L229

yep, we always leak a string if an option is passed.

> https://github.com/ArduPilot/ardupilot/blob/c808ee2f497e7a7a6d570632a3224e655b377323/libraries/AP_HAL_PX4/HAL_PX4_Class.cpp#L274

here, too... but fixing this is not so important.

khancyr, 2016-11-08:
@lucasdemarchi thanks for the review ! It was a little out of my competence !

WickedShell, 2017-02-10:
Closing as this appears to have been fixed for all relevant cases with #5171

closed 2017-02-10 by WickedShell