ArduPilot/ardupilot #5164: sim_vehicle.py can't perform a `param set PARAMETER VALUE` on start

Jesse-Millwood, 2016-11-07:
#### Issue details
When calling the `sim_vehicle.py` script you can't set a parameter on start. 
When trying to set a parameter on start with the `--cmd` switch the output on the console will show 
`Unable to find parameter PARAMETERNAME`

What I have tried: 
```
sim_vehicle.py -d 5 --aircraft Vehicle1 -v ArduCopter --map -m "--state-basedir $HOME/VehicleSims/2Vehicle/ --cmd=\"module load console; param set SYSID_THISMAV 3\""
```
It then shows: 
```
Connect tcp:127.0.0.1:5770 source_system=255
Loaded module map
no script Vehicle1/mavinit.scr
Loaded module console
Unable to find parameter 'SYSID_THISMAV'
Log Directory: /home/cadet/VehicleSims/2Vehicle/Vehicle1/logs/2016-11-07/flight11
Telemetry log: /home/cadet/VehicleSims/2Vehicle/Vehicle1/logs/2016-11-07/flight11/flight.tlog
MAV> Waiting for heartbeat from tcp:127.0.0.1:5770
STABILIZE> Received 689 parameters
Saved 689 parameters to /home/cadet/VehicleSims/2Vehicle/Vehicle1/logs/2016-11-07/flight11/mav.parm
```
**This is also after my change to how the string of `--cmd` is parsed**  #5163 
But you can't pass that to it anyways without my change.

I have tried delaying the start of MAVProxy as well so this might be an issue with MAVProxy.

#### Version
commit: 0c1f2ba144d257fcab069eeeaeca7b2d701953ed

#### Platform
[  ] All
[  ] AntennaTracker
[ x] Copter
[  ] Plane
[  ] Rover

#### Airframe type
Simulated Quadcopter

#### Hardware type
Simulation

#### Logs
_Please provide a link to any relevant logs that show the issue_

khancyr, 2016-11-07:
this is related to mavproxy I think see comment here to make it works: https://github.com/ArduPilot/ardupilot/blob/master/Tools/autotest/sim_vehicle.py#L238

Jesse-Millwood, 2016-11-07:
@khancyr 
Do you know then if there is a way to set parameters on start? 
I tried using the `-A "--defaults /path/to/my/parmfile.parm"` but I don't see it actually doing anything with that argument.

khancyr, 2016-11-07:
@Jesse-Millwood I thought that it was working with param fetch trick... but it doesn't ... I got no solution other than writing it in param file. But it should definitively be improve !

OXINARF, 2016-11-07:
@Jesse-Millwood Like @khancyr said this is an issue with MAVProxy so you should open an issue there. But I doubt there is an easy way to fix this since I believe those commands are run before being connected to the vehicle.

closed 2016-11-07 by OXINARF

Jesse-Millwood, 2016-11-07:
I just made a pull request that allows the user to pass in a parm file to sim_vehicle.py so it will use that instead of the default one. It is #5165

khancyr, 2016-11-07:
@OXINARF It is not only a mavproxy issue. It is also an issue for sitl since we got no easy way to assign parameter on vehicule. For example, if we want to launch a swarm, we can use instance parameter to define all port but it doesn't change mavlink ID , that is annoying and must be solve simpler than loading a parameter file for each uav

peterbarker, 2016-11-07:
mavproxy runs the "startup" commands as soon as sit starts.  Since it hasn't received anything from the vehicle at that point, it has no idea that the parameter exists.

There are a couple of possible fixes:
 - use a mavproxy module that watches the vehicle state and sets the parameter when appropriate
 - write an autotest script instead, using interact() to break out a mavproxy shell
 - Notionally the param module could have some smarts added so that if a param fetch has never ben done then it will defer param sets until one has been done.

OXINARF, 2016-11-07:
@khancyr It isn't a SITL issue as SITL is just an ArduPilot binary made to run with simulated hardware. You can launch as many as instances as you like provided you provide the correct options (including providing a default params file) and you can use whatever GCS you want (to change parameters and everything else).

The sim_vehicle script is just a simple tool to launch the SITL binary and MAVProxy at the same time. A PR with an enhancement to it is never discarded but having an issue opened looks like a stretch to me - otherwise this list becomes even more unmanageable.

peterbarker, 2016-11-10:
@khancyr Changing the vehicle's system id with a param-set currently requires a reboot.  This wasn't a fun thing to discover for my avoidance examples in dronekit-python (https://github.com/peterbarker/dronekit-python/blob/source-system-filtering/examples/avoidance/avoidance-manytests.py#L400)