ArduPilot/ardupilot #5166: Auto land disarm on heli still sometimes failing

tridge, 2016-11-07:
automatic disarm on landing in heli is still sometimes failing. I reproduced it in the RF sim with a Raptor720 today.
http://uav.tridgell.net/Heli/raptor720-disarm-fail.bin
it works fine with LAND mode, but fails with a NAV_LAND in AUTO mode. I think there is some logic bug.

I should be able to look into this myself, but I'm creating this issue so others know about it, especially @lthall and @R-Lefebvre

tridge, 2016-11-07:
note this in the log:
2016-11-08 08:18:18.21: EV {TimeUS : 273415787, Id : 17}
2016-11-08 08:18:20.88: EV {TimeUS : 276086593, Id : 18}
2016-11-08 08:18:20.89: EV {TimeUS : 276094107, Id : 28}
2016-11-08 08:18:21.54: EV {TimeUS : 276749741, Id : 17}
2016-11-08 08:18:24.46: EV {TimeUS : 279667367, Id : 18}
2016-11-08 08:18:24.48: EV {TimeUS : 279686095, Id : 28}
2016-11-08 08:18:25.71: EV {TimeUS : 280914219, Id : 17}
2016-11-08 08:18:30.73: EV {TimeUS : 285936783, Id : 18}
2016-11-08 08:18:30.74: EV {TimeUS : 285948470, Id : 28}

its saying "LAND_COMPLETE_MAYBE", then "LAND_COMPLETE" then "NOT_LANDED"
It just keeps cycling between them continuously. In LAND mode once it gets LAND_COMPLETE state it disarms. In AUTO mode it doesn't - it only checks for LAND_COMPLETE on the next mission update. So it can cycle like this. In this case it kept cycling forever and never disarmed.

rmackay9, 2016-11-07:
We could perhaps add a check into the nav-land to see if there is a next-command (the mission library can tell us that) and then disarm.. or we could call the verify at a much higher rate.

R-Lefebvre, 2016-11-13:
So, this is a "safe" failure right?  ie: the problem is the vehicle is actually landed, but it's not auto-disarming, right?

Is this only a tradheli problem?  Seems that there must be something in the Not_Landed code that is triggering when it should not be.  Sorry if stating the obvious.

ChristopherOlson, 2017-09-26:
Is this still an issue? I'm going thru the issues I find open for TradHeli, tested this one with AC3.5.3 and I can't verify it. The Pixhawk auto disarms fine with a LAND command in AUTO. Unless I'm not testing the right thing.

ChristopherOlson, 2018-09-04:
Just to update this issue, this is being addressed in the spool logic for 3.7. It is not a critical issue and failure to disarm is usually caused by the landing collective min set too low. There is a problem with the logic using throttle_zero instead of limit.throttle_lower that would cause it to cycle from "LAND_COMPLETE" to "NOT_LANDED". WIP PR addresses it
https://github.com/ArduPilot/ardupilot/pull/7894

bnsgeyer, 2019-06-22:
Spool logic was implemented with PR #10656.  Closed.

closed 2019-06-22 by bnsgeyer