ArduPilot/ardupilot #5169: Quadpane QRTL error -strongly shaking

zoucInfo, 2016-11-08:
#### Issue details
At first I did an auto-fly plan(Attachment:Flight Plan.txt). Everything was working normal as expecting. But something went wrong when it entered into QRTL.
QuadPlane is violent shaking when in QRTL mode(TIME:2016-10-20 16-16-31 16:19:53 Attachment:2016-10-20 16-16-31.tlog AND Error Log.log). At this time ,the ground-speed is 25m/s, barometer height is 150m and only “bad compass health” prompting. Shown as below.
#### Version
Plane3.7

#### Platform
[  ] All
[  ] AntennaTracker
[  ] Copter
[ x ] Plane
[  ] Rover

#### Airframe type
quadpane

#### Hardware type
pixhawk

#### Logs
2016-10-20 16-16-31.tlog [2016-10-20 16-16-31.tlog](http://discuss.ardupilot.org/uploads/default/original/2X/9/9df9ca9e16df7453a46b44317f3ff13a05acfe8a.tlog)
Error Log.log[Error Log.log](http://discuss.ardupilot.org/uploads/default/original/2X/e/e992f6bc86ad8be471bc93c4bae6ab037542bef3.log)
Flight Plan.txt [Flight Plan.txt](http://discuss.ardupilot.org/uploads/default/original/2X/e/e0ae53c095a9c86cfaa502af9e38ec5b055a8a20.txt)

OXINARF, 2016-11-08:
@zoucInfo Since you already posted I'll leave it open for Plane devs to look at it, but this is a support question and those should be posted at the forum (http://discuss.ardupilot.org), so next time do that.

Naterater, 2019-01-05:
Please post to the support forum until it can be guaranteed that there is a flaw in the code (and not the aircraft or tune).

closed 2019-01-05 by Naterater