ArduPilot/ardupilot #5173: Issue with ubuntu prereqs.sh script

Autonomost, 2016-11-09:
#### Issue details
When trying to run "ardupilot/Tools/scripts/install-prereqs-ubuntu.sh -y" 

I get the error "+ sudo pip2 -q install -U future lxml pymavlink MAVProxy
Command "python setup.py egg_info" failed with error code 1 in /tmp/pip-build-s2ZPBw/pymavlink/" 

#### Version
Newest version

#### Platform
[X ] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover

#### Airframe type
None

#### Hardware type
Pixhawk

#### Logs
_Please provide a link to any relevant logs that show the issue_

OXINARF, 2016-11-09:
We are aware of it and trying to fix it. It isn't a problem with ArduPilot but with pymavlink. For now you can do `pip install future lxml` and then run the script again.

Autonomost, 2016-11-09:
thank you that fixed it!

peterbarker, 2017-12-19:
This was fixed.

closed 2017-12-19 by peterbarker