ArduPilot/ardupilot #5175: Delete old dev branches

magicrub, 2016-11-09:
Goal: delete extra [ardupilot/ardupilot](https://github.com/ArduPilot/ardupilot) git branches that are not needed for archive or active development work.

- [ ] [mavlink2-wip](https://github.com/ArduPilot/ardupilot/tree/mavlink2-wip) - This has conflicts and generally appears to be old. Last commit was April 4, 2016
- [ ] [px4-merge-april-2016](https://github.com/ArduPilot/ardupilot/tree/px4-merge-april-2016) - all commits are merged into master
- [ ] [uavcan_update2](https://github.com/ArduPilot/ardupilot/commits/uavcan_update2)
- [ ] [uavcan_update3](https://github.com/ArduPilot/ardupilot/commits/uavcan_update3) - 2 commits ahead of update2 and they are all merged into master
- [ ] [pymavlink-update](https://github.com/ArduPilot/ardupilot/commits/pymavlink-update) - all commits are merged into master

OXINARF, 2016-11-09:
I think mavlink2-wip can be deleted safely, I think there was another mavlink2 branch after that one.

magicrub, 2016-11-09:
That's what I'm thinking but it would be nice for the author of that branch @tridge to confirm before we do any destructive operations. No hurry

OXINARF, 2017-07-05:
I'm not sure who did it, but mavlink2-wip branch has already been deleted. Closing.

closed 2017-07-05 by OXINARF