ArduPilot/ardupilot #5176: Delete ArduCopter version branches in favor of tags

magicrub, 2016-11-09:
delete copter specific version branches in favor of using tags which plane, tracker, and rover already use.
ArduCopter-2.8.1
ArduCopter-2.9
ArduCopter-3.0
ArduCopter-3.1
ArduCopter-3.1.1
ArduCopter-3.1.2
ArduCopter-3.2
ArduCopter-3.2.1
Copter-3.3
Copter-3.3-pixracer
Copter-3.4

We'll want to replace that with a new branch called ArduCopter-release

billbonney, 2016-11-09:
You have an option to create an 'Archive' repository. i.e. one that that is clone of ardupilot called ardupilot/ardupilot-archive and then delete the branches from the main repo. This at least keeps place to keep the history, and for developers with projects based on these older variants to find. Please don't delete them unless they have been copied and kept safe elsewhere.

You may still need to create new release branches, e.g copter-3.4.1 etc, even if you have a main 'copter-release' branch. If you need to patch a release and you want that to be minimal it is the only way.

Is summary, the solution is to archive branches to cloned repo and then delete them from the main development repo.

OXINARF, 2016-11-09:
I don't think another repository is a good idea. As @billbonney has experienced a tag is very similar to a branch, so deleting the branch leaves it there if it has a tag.
I like to have few branches so I like the idea of having tags instead of branches but I'm preparing a document with a few rules for releases and that includes what to do in GitHub, so I'd like to wait before going ahead with this.

magicrub, 2016-11-09:
Bill, these are all tagged so deleting the branch won't lose anything.

billbonney, 2016-11-11:
> As @billbonney has experienced a tag is very similar to a branch
So name pollution in the tag space is ok, but in the branch space it's not.  I don't see the difference. I'm also not that bothered by the branches. we can create new branches and just move on. 

It does impact the work flow not having branches, since if i create a fix to a release (which would require checking out a commit, make in a branch and then try and submit a PR, I can't as the PR cannot be raised against a TAG (even though the text in the entry box seems to indicate you can, you can't)

[image]

I think having a separate archive repo for the old branches and just keep the new main repo is an interesting idea. you can remove the old branches/tags (and history) reducing the repo size. But I'm not advocating that happen. it was just a solution to what @magicrub proposed.

magicrub, 2016-11-12:
branches for PR's need to merge with master without conflicts, ideally they were recently rebased onto current master. It's silly to consider a PR based on copter v3.1.2 tag or branch.

OXINARF, 2016-11-12:
> So name pollution in the tag space is ok, but in the branch space it's not. I don't see the difference.

Tags are more hidden in GitHub and almost all Git UIs. Having it as tags removes more distraction.

> It does impact the work flow not having branches, since if i create a fix to a release (which would require checking out a commit, make it a branch in my fork and then try and submit a PR to upstream, I can't as the PR cannot be raised against a TAG (even though the text in the entry box seems to indicate you can, you can't)

I agree that can be a problem, although it's not too common to see PRs for old versions (the latest version would always have a branch). We can have the two latest versions, I don't see we have the resources to be supporting more than that.

> I think having a separate archive repo for the old branches and just keep the new main repo is an interesting idea.

In my opinion that just causes more confusion and it increases maintenance work.

> branches for PR's need to merge with master without conflicts

If it is a fix for an older version it may not apply to master at all.

Naterater, 2019-01-08:
Do we still want to do this or has this been done yet?

OXINARF, 2019-01-09:
We still want this.

rmackay9, 2024-03-18:
I might close this if that's OK.  I like the current method of using version specific release.  Rover, Plane and even Tracker have adopted the Copter method now as well.

closed 2024-03-18 by rmackay9