ArduPilot/ardupilot #5191: Incorrect 'No Terrain Data' message sent in Auto mode with no GPS

mboland, 2016-11-11:
#### Issue details
If an auto mode is selected without an adequate GPS lock a 'FS: No Terrain Data' message is sent even if terrain use is turned off and the mission in memory has not used Terrain for height.

#### Version
AC3.4.x

#### Platform
[  ] All
[  ] AntennaTracker
[ X] Copter
[  ] Plane
[  ] Rover

#### Airframe type
Octocopter +

#### Hardware type
3DR Pixhawk, Ublox GPS, Septentrio AsteRx-m UAS GPS

#### Logs
http://uavs.com.au/Arducopter/logs/2016-11-11

OXINARF, 2016-11-11:
Can you post a screenshot when you get this error? I looked at your log, I looked at ArduPilot code, I looked at Mission Planner code and I couldn't find anywhere that this message is generated.

WickedShell, 2016-11-11:
@OXINARF 
https://github.com/ArduPilot/MissionPlanner/blob/ed9320a8fcc7ddb5ef32363f292d7be8b2599fb2/CurrentState.cs#L1640
https://github.com/ArduPilot/MissionPlanner/blob/1f9193f5c091a2d40001e9c644a224566496121d/Strings.resx#L507

OXINARF, 2016-11-11:
@WickedShell I saw that but the message Mike indicates (several times in the forum and here) is different. It has FS as a failsafe has occurred.

mboland, 2016-11-12:
Won't be back on deck for a week and this copter is now a pile of parts on the bench until I get back.

This error should be easy to duplicate, even with a Ublox, just fire up the copter indoors, and without a GPS lock switch to auto with a mission in memory. No need to arm.

It generates the message, viewable in messages on the GCS, immediately.

OXINARF, 2016-11-14:
According to http://discuss.ardupilot.org/t/auto-terrain-following/12832 the message is “Failsafe: Terrain data missing” and that I can find. I'll try to debug this this week.

chwieser, 2016-11-19:
I had the same issue again and you can find the logs at: https://drive.google.com/drive/folders/0B-lDc9-zD0TFQW5sSFlBSHh4dmc?usp=sharing

hope that helps.

Krisu

rmackay9, 2016-11-21:
@chwieser, I think the terrain failsafe you're seeing has a different cause.  It looks like maybe you're using a Lidar Lite with the PWM interface and although that lidar has a range of about 40m the RNGFND_MAX_CM is set to only 7m.  So I think that should be increased to perhaps 3700 (i.e. 37m).
This is mentioned on the wiki in the "Setup in Mission Planner" although I can see how you might skip over that if you thought it was already covered in the "Connecting to Pixhawk via PWM".
http://ardupilot.org/copter/docs/common-rangefinder-lidarlite.html#setup-in-mission-planner

rmackay9, 2016-11-21:
Re this issue, I think it's all resolved in AC-3.4.2 (fixes appear as "SBF GPS altitude and accuracy reporting fix" in the release notes) so I'm going to close this item.  We can re-open if @mboland says that AC3.4.2 isn't working.

closed 2016-11-21 by rmackay9

chwieser, 2016-12-07:
@rmackay9, thank you - I can confirm that after setting RNGFND_MAX_CM to 4000 the issue disappeared. Terrain following works now like a sharm :)