ArduPilot/ardupilot #5197: Arduplane 3.7.1 compatibility issue with GRXX graupner receiver through SUMD interface

Benoit3, 2016-11-11:
#### Issue details

I had some problem to have a GR16 working with Pixhawk and ArduPlane 3.7.1. It was OK with a GR12L, but not with a GR16.

I want to use a GR16, even more expensive, to be able to use this mavlink to hott converter

with all features.

For the GR16 I use the SUMDOF08 setting.

But I was not able to get the Pixhawk out of Failsafe.

After a dive into the SUMD protocol specification, after having analysis GR12L and GR16 output signal with a ftdi cable 'usb to serial converter) and a serial analyser (moserial under linux) I found a difference between GR12L and GR16 :

At power up, GR12L doesn't transmit any frame up to Tx/Rx link establishment

At power up, GR16 transmit SUMD frame with Failsafe SUMD Header up to Tx/Rx link establishment. After this establishment, if the link is broken, the GR16 doesn't transmit any frame.
As all GR XX (without L) receivers seems to have similar software, this issue exists probably too for GR 12, GR 24 and GR 32
With these information, I dig into the pixhawk code, and found the ardupilot/modules/PX4Firmware/src/lib/rc/sumd.c and sumd.h files. These files belong to the PX4Firmware module.
A quick look into these files shows that SUMD frames with failsafe header are not recognized as SUMD frames. I don't exactly how protocol detection works, but it seems that the system doesn't recognize the SUMD protocol and disable the dsm input to the next reboot.

#### Version
ArduPlane 3.7.1 

#### Platform
[  ] All
[  ] AntennaTracker
[  ] Copter
[X] Plane
[  ] Rover

#### Airframe type
flying wing

#### Hardware type
PixHawk V2

#### Logs

Naterater, 2019-01-08:
Is this still an issue with recent firmware @Benoit3?

Benoit3, 2019-01-10:
Hello Nathan,

I didn't have any return of my proposal for 2 years. I don't think it has been merged in the ardupilot trunk. So if the bug has not been corrected with another solution, it should be still there.

Due to lack of time, I didn't fly my plane since a little more than one year. I'm not aware of a new version which could correct it.
Regards,
Benoit

IamPete1, 2021-08-30:
This issue seems to be related to the old PX4 hal, so i'm going to close.

closed 2021-08-30 by IamPete1