ArduPilot/ardupilot #5210: Copter: throttle jump when EKF switches lanes

rmackay9, 2016-11-15:
Copter-3.4.1 (and higher) has two EKFs running simultaneously.  When we switch the throttle can jump momentarily (perhaps the earth frame accelerations are jumping?).  The EKF notifies us when a lane switch has occurred (see check_ekf_reset function) so when we receive this notification we should "freeze" the D term for the next iteration of the altitude controller.

rmackay9, 2016-11-21:
I've asked Paul if he can provide the change in estimated altitude change similar to what we have for the estimated position change (i.e. "getLastPosNorthEastReset").

priseborough, 2016-11-21:
I will start on the vertical position reset today. I have also submitted a PR to reduce the likelihood of unwanted EKF core switches here: https://github.com/ArduPilot/ardupilot/pull/5252

rmackay9, 2017-03-21:
This is included in AC3.5 now so closing.  Because only positions (and maybe velocities) are reset, it's actually possible that there will be a throttle "twitch" but it should be small.

closed 2017-03-21 by rmackay9