ArduPilot/ardupilot #5218: Current sensing with Pixracer seems broken (it always measures 1 V at the current pin)

dgrat, 2016-11-15:
#### Issue details
I use a battery monitor which gives me something like ~0.05-0.1 V at the current pin (estimated by volt-meter).
However, with a divider of e.g. 1 Amp/Volt  I get always 1 Amp.  
The reading of the pin is therefore always equal to the divider. 
The voltage pin works as expected and equals the volt-meter reading.

#### Version
3.4

#### Platform
[  ] All
[  ] AntennaTracker
[ X] Copter
[  ] Plane
[  ] Rover

#### Airframe type
Quad

#### Hardware type
Pixracer

#### Logs
No logs

lvale, 2016-11-15:
I have a PixRacer from AUAV.co and I've used also their ACSP5 monitor http://www.auav.co/product-p/acsp5.htm and also a Mauch http://www.mauch-electronic.com/apps/webstore/products/show/6626769
and never had a problem

dgrat, 2016-11-16:
Thanks for reply. Good to know this. I bought one from ebay and it behaves somehow odd in comparison to the Navio+/2 which I also use. Beside to this weird problem the RSSI signal is not steady, and the servo is sometimes not starting like I am used to. Additionally, arming only works if I connect the Pixracer to USB. Just powering from battery does not work. Even though the voltage is a perfect 5.3V line. I also don't like the arming switch. Now I am in doubt to really use this board for a flight test because I am afraid it will just crash the model. I also have no clue where the 1V signal comes from. The Mauch module delivers the expected voltage and the cable was also checked.

dgrat, 2016-11-16:
I rewired. The current sens wire goes now to the voltage pin. 
Now I can sens the current but not the voltage.
I can make clear at this point that either my Pixracer is broken or it is a problem in software.

lvale, 2016-11-17:
I would bet that something is broken on your setup.

Closing the issue for now.

closed 2016-11-17 by lvale

dgrat, 2016-11-17:
@lvale 
I had never a PCB which constantly has a voltage on a (unconnected) pin without input. 
Even if nothing is connected it says ~0.5V :/ 
I think it is just crap hardware and there is a short somewhere. 
I just wonder whether I should directly buy something different.

R-Lefebvre, 2016-11-19:
Is this a genuine AUAV Pixracer or a clone?

dgrat, 2016-11-19:
@R-Lefebvre 
When I bought, I thougt it is an original.
Now I am 90% sure it is a clone :/

magicrub, 2016-11-21:
@dgrat where did you buy it from?

dgrat, 2016-11-22:
@magicrub 
I bought it here. Maybe someone can validate whether copy or not.
Could also be that it was just bad luck, to get a strange unit. 

http://www.ebay.de/itm/ws/eBayISAPI.dll?ViewItem&item=272330306347&ssPageName=ADME:X:RRNTB:DE:3160

OXINARF, 2016-11-22:
If you didn't buy from AUAV.co (or mRobotics that is now selling it too) then most certainly it is a clone. Why would you believe that an item sold in eBay at almost half the price of the original and from a user called *trustinchina* would be an official product?

lvale, 2016-11-22:
It's a copycat of the PixRacer, so anything can happen....

R-Lefebvre, 2016-11-22:
Great example of why I believe we should try to work with legitimate hardware vendors to provide locks to prevent usage of clones of questionable quality.  We have to remove the market for these things.

magicrub, 2016-11-22:
You get what you pay for

R-Lefebvre, 2016-11-22:
The problem is people losing their money on these things that are not just clones, but they counterfeit the trademark too.  So people like Daniel don't even know what they are getting.

magicrub, 2016-11-22:
Should I be mad that my Rolodex that I got from eBay for 10% the usual price ended up being fake? Do I blame Rolodex for that?

OXINARF, 2016-11-22:
@R-Lefebvre I can agree with you in certain situations, but not when you buy something from eBay.

dgrat, 2016-11-23:
In my case, the board was relatively close to the original price ~70€ and @AUAV it was sold out :/
Despite the fact that I was not lucky, i think banning clone boards is a bad idea and there are some nice clones out.
However, I suggest maybe to implement a warning at the GCS if some clones are known to cause trouble and if it can be verified.
Actually, I just opened this issue in case there might have been a software issue.

matt88, 2018-04-03:
Did you ever get this resolved? I have the exact same issue on a pixracer and my pixracer does have auav stamped on the board. I have another pixracer I purchased from auav directly and the boards are identical. No way this is a clone. Either way I’d like to get this resolved because the board works flawlessly otherwise. I’ve ran it in a tarot 680 pro for two flying seasons without any major issues.

utkinpol, 2018-06-06:
i got pixracer from mRo store just recently. got ACSP4 board. it is connected straight into the FC with the provided cord. 
options set as monitor - 4.voltage and Current, Sensor: 0.Other, APM ver: 4. The Cube or Pixhawk.

I get voltage - do not get current. starting hating all this, i cannot find what is wrong. then found this thread. Will try to send email to mRo, may be they can advice, it is odd.