ArduPilot/ardupilot #5221: SITL with x-plane does not connecting mission planner

Abdullah1990, 2016-11-16:
#### Issue details
As I Run the SITL using mission planner with the x-plane selected as model type. The arduplane.exe wont connect to the ground station. As if it's not sending any heartbeats to the ground station. I tried the SITL with the PLANE it works fine. It seems that when the Arduplane.exe connects with x-plane thru UDP it wont connect to the ground station  (Mission planner 1.3.42) using TCP.

-- If a run the Arduplane.exe from command line with parameter --model PLANE  it works fine. 
but if I run it with --model XPLANE mission planner can not connect to mission planner

-- The TCP  port to connect to missiion planner is port # 5760
#### Version
The SITL that is used Arduplane v3.8 (Beta1 (c0303ffe))

#### Platform
[  ] All
[  ] AntennaTracker
[  ] Copter
[ x ] Plane
[  ] Rover

#### Airframe type
Fixed win
#### Hardware type

ayoubzar, 2017-03-09:
did you finnd any solution I am facing the same problem again ???

DawoodTaii, 2018-02-22:
By experience, just make sure that x-plane is running before connecting to SITL. Once x-plane is started with correct configuration there should be no problem.

IamPete1, 2021-08-15:
Sorry we never got to this.

closed 2021-08-15 by IamPete1

Abdullah1990, 2021-08-15:
That is ok, I think it has been fixed.