ArduPilot/ardupilot #5225: Automatically updating firmware on UAVCAN connected nodes

pavel-kirienko, 2016-11-17:
At the moment, it is not possible to upload new firmware to a UAVCAN connected node from an APM-powered autopilot.

PX4 has this feature since version 1.0; it is *somewhat* documented here <http://dev.px4.io/uavcan-node-firmware.html#firmware-installation-on-the-autopilot>, and the implementation is located here: <https://github.com/PX4/Firmware/blob/master/src/modules/uavcan/uavcan_servers.cpp>.

Protocol-level documentation is available at <http://uavcan.org/Specification/6._Application_level_functions/#firmware-update>, and Libuavcan-specific docs are at <http://uavcan.org/Implementations/Libuavcan/Tutorials/11._Firmware_update>.

gmorph, 2016-11-17:
Which UAVCAN devices are being used to test this and have new firmware uploaded?

pavel-kirienko, 2016-11-17:
* [Zubax GNSS 2](https://docs.zubax.com/zubax_gnss_2)
* ESC based on PX4 Sapog ([Zubax Orel](https://docs.zubax.com/zubax_orel_20), Pixhawk ESC v1.4)
* Thiemar S2740VC

Vishranthee, 2019-12-09:
Is there any update on this issue? Facing the same problem .

IamPete1, 2021-08-30:
We can now update UAVCAN stuff via SLCAN, https://ardupilot.org/copter/docs/common-uavcan-setup-advanced.html#slcan

closed 2021-08-30 by IamPete1