ArduPilot/ardupilot #5226: Navio start crash

samirbonnet, 2016-11-17:
#### Issue details
SIGSEGV, Segmentation fault on startup with Navio+ on RPI3

#### Version
master (11/17/2016)

#### Platform
[  ] All
[  ] AntennaTracker
[  ] Copter
[ X ] Plane
[  ] Rover

#### Airframe type
Fixed Wing

#### Hardware type
Navio+ on RPI3

#### Logs

[New Thread 0x76bf1400 (LWP 1115)]
[New Thread 0x763f1400 (LWP 1116)]
Raspberry Pi 2/3 with BCM2709!

Program received signal SIGSEGV, Segmentation fault.
0x000e4b54 in AP_ADC_ADS1115::AP_ADC_ADS1115() ()
(gdb) bt
#0  0x000e4b54 in AP_ADC_ADS1115::AP_ADC_ADS1115() ()
#1  0x000e20cc in AnalogIn_ADS1115::AnalogIn_ADS1115() ()
#2  0x00013078 in _GLOBAL__sub_I__ZN9HAL_LinuxC2Ev ()
#3  0x000ea518 in __libc_csu_init ()
#4  0x76c7f228 in __libc_start_main (main=0x7efff7f4, argc=1994014720, 
    argv=0x76c7f228 <__libc_start_main+168>, init=0xea4c8 <__libc_csu_init>, 
    fini=0xea52c <__libc_csu_fini>, rtld_fini=0x76fdf408 <_dl_fini>, 
    stack_end=0x7efff7f4) at libc-start.c:246
#5  0x00013530 in _start ()

OXINARF, 2016-11-17:
@lucasdemarchi Do you have a Navio+ to test this?

lucasdemarchi, 2016-11-17:
I do. I will test it today

lucasdemarchi, 2016-11-18:
This crash is fixed by #5235

closed 2017-05-12 by lucasdemarchi