ArduPilot/ardupilot #5242: AP_Baro: Wrong source for the air-temperature results in huge alt-errors

Code-Tester, 2016-11-20:
#### Issue details
There is still a wrong altitude calculation in the Baro-Code. We can't take the ground temperature from the baro to calculate the altitude.
Instead we must use an external air-temperature sensor outside of the pixhawk and the frame.

Please let me explain the problem with an example:

Outside air temperature at the ground: 5 Degree Celsius
Ground Pressure: 1013,25 hPa
Pressure in the air: 1001,20 hPa

Calculated baro_alt: 97,29 m above the the ground.

When we use 35 Degrees Celsius for the ground temperature with the same pressures than we get a baro_alt of: 107,78 m.

The error is greater than 10 m in the altitude.

What's happened:

We go outside with the Copter at a ground temperature of 5 Degree Celsius (air temperature).
Okay let's wait for 30 minutes to conditioning all parts inside of the pixhawk especially the Baro. Now the Baro has approximately the same temperature as the ambient air temperature.

Time for booting... The Firmware will fix the first temperature readings as the ground temperature. Perhaps our ground temperature is now okay but defently not accurate.

Let's fly for a while. Our Pixhawk heats up and also the baro. And now we have a great problem, when we change the battery pack, the pixhawk will boot again and fix a very high ground temperature.

We should use the baro-temperature-readings only for compensation of the temp and pressure readings (you can see this at the datasheet).

But for the altitude calculation we need an external accurate air-temperature sensor.

Okay we can also use the Airspeed-Sensor for this, but when you use a COPTER the Sensor is almost useless. Because you can measure the Airspeed only in one direction....

So when we use a COPTER-Frame it's better to use a standalone air-temperature sensor outside of the frame. The combination of air-temperature and air-humidity will be perfect.

With the air-humidity you can also calculate the exact gas-constant and the air-density.

Hey Guys, what do you think? Who of you can help me to create a new driver for a standalone air-temperature and air-humidity sensor over I2C...

1 Step: We must find a good sensor. Suggestions?
2 Step: I will buy this sensor
3 Step: I need help to implement the sensor protocol and to change the temp-source inside of the baro code. (I2C-Driver and so on)
4 Step: I will be the Beta-Tester with real flights.
5 Step: Official Release for the Community and for the Docs.

Thanks in advance for the help. I think this issue is really important because not everybody from us will use a rangefinder for correct readings...

#### Version
V3.4.2

#### Platform
[  ] All
[  ] AntennaTracker
[x] Copter
[x] Plane
[  ] Rover

#### Airframe type
Quad

#### Hardware type
Pixhawk V1 and Pixhawk V2

#### Logs
no

khancyr, 2016-11-20:
Recommendation MCP9808, accurate and simple and hightly accessible for diy communauty

Code-Tester, 2016-11-20:
Another Idea is to use the AM2315. Here you can have a look at the specs: https://www.adafruit.com/products/1293

But at first we need to implement the I2C driver.... for that sensor.

WickedShell, 2017-11-03:
Baro temp handling was reworked quite a lot in #5930, external/generic temperature sensing is already being tracked in #1799

closed 2017-11-03 by WickedShell