ArduPilot/ardupilot #5244: feature request: failsafe by throttle level

night-ghost, 2016-11-21:
#### Issue details
as the battery discharge, hovering throttle gradually increases. But when the hovering throttle comes over 70% copter becomes difficult to stabilize and increases the likelihood of crash due to turbulence. With bad battery this can occur before voltage failsafe.
Therefore, it seems logical to add a parameter which triggers failsafe with a choose of mode when a certain hovering throttle level is reached.

#### Platform
[  ] All
[  ] AntennaTracker
[x] Copter
[  ] Plane
[  ] Rover

rmackay9, 2016-11-21:
I'm really not sure this would be the correct way to do it.  Instead it would be better to set the existing battery failsafe so that the voltage is high enough that the vehicle can maintain a reasonable throttle level.

night-ghost, 2016-11-21:
There was a flight when throttle level was a 77% on takeoff without triggering a voltage failsafe at 10.5 (3s). I was able to put copter by miracle - he flew like a drunk, swaying.

rmackay9, 2016-11-21:
@night-ghost, ok.. there was another suggestion I think to have a separate take-off battery voltage arming-check to protect against people taking off after forgetting to charge the battery.  It's possible though that this kind of thing should be avoided by improvements in the ground station to show the batter level more clearly.

night-ghost, 2016-11-21:
Oh no, I don't forget to charge battery! This was a new fully charged battery - but very bad battery with internal resistance 0.1 Ohm. And this was a test fly so neither telemetry nor OSD.

Prearm battery check is a good idea too, but it has different aims.

lvale, 2016-11-21:
@night-ghost @rmackay9 
The issue @night-ghost refers can be handled by using the MOT_ parameters http://ardupilot.org/copter/docs/parameters.html#mot-bat-volt-max-battery-voltage-compensation-maximum-voltage

and

http://ardupilot.org/copter/docs/parameters.html#mot-bat-volt-min-battery-voltage-compensation-minimum-voltage

wjax, 2016-11-22:
@night-ghost Battery failsafe is the way to go. It triggers based on voltage.
Introducing another failsafe tailored to a very specific situation and copter configuration can only add more noise and complexity.
Voltage sag will make the battery failsafe to kink in. Just be sure to set battery failsafe correctly. Another thing should be take into account and it is the correct copter design as to have a good amount of power available even with a battery at low capacity and windy condition.

Anyway and as a side note, battery voltage takes some good moment to kick in. I understand that it waits for a given time to be under defined voltage. However every time I test it, it takes around 20 secs from the moment mission planner shows battery levels below threshold.

R-Lefebvre, 2016-11-22:
Agree with Luis.  Those parameters can help.  Did you use them.

I'm concerned this change would drive up complexity to protect for a rare situation which is really a vehicle system design or operations problem.

Naterater, 2019-01-08:
I think this issue should close. Voltage dropping is the normal cause of increase throttle level. Based on the dynamic nature of missions, I don't think throttle values are a good indicator that a failsafe should be triggered.

rmackay9, 2019-01-09:
The hover throttle is only updated when the vehicle is flying level so we would probably find that it simply doesn't update during a long mission which doesn't have periods where the vehicle is hovering.

I think we do have problems still with people crashing because they've run out of batteries but it's generally because of a mistake in the setup (i.e. they've set the voltage too low).  We've added the optional pre-arm voltage check (using the ARMING_VOLT_MIN check).  My feeling is the next step to improve our battery failsafes is to add a calculation of what capacity we need to get home.

So in the end, although I think this issue has a good-ish idea in it, I think we will tackle the issue with other checks.

closed 2019-01-09 by rmackay9