ArduPilot/ardupilot #5246: feature request: don't turn off motors on high angles

night-ghost, 2016-11-21:
#### Issue details
When the angles of inclination of copter reaches a certain value, the controller ceases to struggle on and turns off motors. I think it is wrong, and when inertial system feels motion, controller should try to stabilize the copter. As in this video:
https://www.youtube.com/watch?v=HvZqmtOm-dA

#### Platform
[  ] All
[  ] AntennaTracker
[X] Copter
[  ] Plane
[  ] Rover

rmackay9, 2016-11-21:
The vehicle will always try to right itself but the crash check may trigger if the vehicle's desired and actual attitude are off by more than 30 degrees for two uninterrupted seconds.  This feature can be disabled though: http://ardupilot.org/copter/docs/crash_check.html

I'm going to close this item because I think it's already handled.

closed 2016-11-21 by rmackay9

magicrub, 2016-11-21:
I'd like to add that I think the behavior on that video is dangerous. If it hit something and is actively chewing into it then it should halt.

night-ghost, 2016-11-21:
crash check is a different thing! Copter firmware makes a crash when I change mode from Acro to Stabilize  in upside down position.

rmackay9, 2016-11-21:
@night-ghost, it should definitely recover.. we can test in the simulator.  any chance you have a log file?

night-ghost, 2016-11-21:
My testing copter is (almost) indestructible so I'll retry this in a few days and write logs.

PS. May be things changed somewhen, this was on 3.2 firmware.

R-Lefebvre, 2016-11-22:
I've done this A LOT with with AC3.2 and 3.3 on tradheli, and some testing with 3.4.  I've never had a problem changing from Acro to Stabilize.