ArduPilot/ardupilot #5256: Copter: move run_rate_controller before ahrs.update to reduce delay

rmackay9, 2016-11-22:
We can reduce the latency in our rate loops (which are arguably the most critical) by shifting the attitude_control.run_rate_controller() to run before the read_AHRS().
https://github.com/ArduPilot/ardupilot/blob/master/ArduCopter/ArduCopter.cpp#L262

The only issue is that we need a new method in our AHRS library to return the latest ins.get_gyro() information but also include the the gyro bias estimate from the previous iteration.  this new method should also use the AHRS/EKF's decision on which IMU's gyro data is more reliable.

Inside our rate controllers (see below) we should then replace the ahrs.get_gyro() call with this new get_gyro_latest().
https://github.com/ArduPilot/ardupilot/blob/master/libraries/AC_AttitudeControl/AC_AttitudeControl.cpp#L591

priseborough, 2016-11-22:
When running both lanes, the EKF typically takes anything from 500uSec to 2000 uSec to run depending on interrupts and the measurements that need to be fused.

rmackay9, 2017-05-10:
This has been completed and is included in Copter-3.5-rc4

closed 2017-05-10 by rmackay9