ArduPilot/ardupilot #5258: Handling Custom Rotation Matrix for sensors

jberaud, 2016-11-22:
The real rotation for the Compass on Bebop 2 is not an exact one and I would like to be able to use the following matrix:
{{ -0.984807753012208,  0, -0.173648177666930},
  {                  0,               -1,    0},
  { -0.173648177666930,  0,  0.984807753012208}}

jberaud, 2016-11-22:
Also I don't think that we should add all custom rotations in vector3.cpp but rather have a generic way of handling them and define them in the hal description.

magicrub, 2016-11-22:
How about the Matrix3 class? https://github.com/ArduPilot/ardupilot/blob/master/libraries/AP_Math/matrix3.h

tridge, 2018-03-06:
how far off one of the standard rotations is this? Perhaps we need something equivalent of AHRS_TRIM_X and AHRS_TRIM_Y but for compasses?

KhushalB, 2018-09-21:
Any update on this? I would also like support for custom pixhawk and compass orientations.

IamPete1, 2021-07-25:
closed by https://github.com/ArduPilot/ardupilot/pull/13586

closed 2021-07-25 by IamPete1