ArduPilot/ardupilot #5272: PH2: UAVCAN GPS not found on CAN1

WickedShell, 2016-11-24:
#### Issue details
UAVCAN GPS (Zubax GNSS2) is not detected on CAN1. Confirmed with AC3.4.2, but I suspect it applies to all firmware versions and have tagged it as such. Tested with BRD_UAVCAN 2, and GPS_TYPE 9. (Power cycled after shifting ports as #4902 is still unfixed).

#### Version
AC3.4.2 (bed521e2b57e706)

#### Platform
[ x ] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover

#### Hardware type
Pixhawk 2

WickedShell, 2016-11-24:
@tridge if you get a chance to look at this that would be helpful. I've confirmed that plane has the same behaviour (44c5fee90d87)

benjammin1988, 2016-11-24:
Same issue here, except I still cannot get external compass detection on can 2, only GPS

benjammin1988, 2016-12-10:
Issue not being able to detect external compass still persists on 3.4.3

proficnc, 2016-12-10:
Is yours a version 1 or 2?

benjammin1988, 2016-12-10:
Pixhawk 2

proficnc, 2016-12-10:
I meant the GPS :).  What version is that?  Tridge has verified version 1 on his desk... but none of us have version 2 etc...

benjammin1988, 2016-12-10:
It's version 2.  If you want I can send it to Tridge.

Let me know!

Ben

proficnc, 2016-12-10:
I will ping you offline

WickedShell, 2016-12-10:
I verified with Zubax 2 originally. Out of the country for a week though...

benjammin1988, 2016-12-10:
Did you verify compass detection or just GPS?

WickedShell, 2016-12-10:
At the time I was only verifying GPS.

benjammin1988, 2016-12-10:
I tried on 3 different units to make sure it wasn't a bad zubax compass. GPS gets satellites no problem, but APM will only use the internal compass.

magicrub, 2016-12-11:
This issue is for CAN GPS, which is working. Let's open a new issue for CAN
compass.

WickedShell, 2016-12-11:
I almost guarantee its the same issue, I'd suggest keeping one issue for
now.

benjammin1988, 2016-12-11:
Idk if it's necessary to add this to the issue, but can 1 also won't recognize the GPS/compass altogether.  The GPS is only recognized through can 2.  Neither recognize the external compass and only can 2 recognizes the GPS

RoyPercepto, 2017-01-22:
Hi,
Any updates regarding this issue?
I can't get GPS/Compass on CAN1 and on CAN2 no Compass

benjammin1988, 2017-01-22:
Not as far as I know..  I am willing to send whoever a zubax unit to get this fixed.

magicrub, 2017-01-26:
@benjammin1988 This is a generous offer, thank you! Any takers?

tridge, 2017-01-26:
@benjammin1988 the most common issue is slow startup times of the CAN device, leading to it not being available when sensors are probed. I increased the delay time waiting for CAN startup recently. Can you please try with master (ie. 'latest' in MissionPlanner). 
If it doesn't work, then please get a nsh shell, and run "uavcan status" first with the zubax on CAN1 then on CAN2. There are two ways to get a nsh shell:
1) boot with no microSD inserted and connect to USB port as a serial device. Then run "uorb start", "uavcan start" and then after about 10 seconds run "uavcan status"
2) boot normally and use nsh over MAVLink. MissionPlanner has this in its terminal connection menu. Just run "uavcan status" as uavcan will already be started

I'd love to get a Zubax2, but I suspect we'll be able to solve this together without you sending me one. For realtime diagnostics together please try gitter https://gitter.im/ArduPilot/Hardware or ping me on Skype as AndrewTridgell (or both!)

benjammin1988, 2017-01-27:
Tridge, I'm aware and have been following the issue for quite some time... Your delay did help zubax gnss 2 search for satellites on can 1, but still no compass detection on either can 1 or can 2. 

I don't know when I'll have time to do your fixes as I am managing a fleet of spray drones for a partner company and my own company at the same time. When the issues arised, I had to quickly switched to drotek's m8n gps.

Does anyone else following want to try these for us? If not, I will gladly send you a zubax gnss 2 as we put them back on the shelf.

Regards,

Ben

davidbitton, 2017-02-04:
I have a PH2.1 and a Zubax 2. Here are the results from the nuttshell:

```
NuttShell (NSH)
nsh> uorb start
nsh> uavcan start
PX4: param_find(UAVCAN_NODE_ID)
PX4: param_find(UAVCAN_BITRATE)
uavcan: Node ID 1, bitrate 1000000
uavcan: SW version vcs_commit: 0x8a196281
uavcan: sensor bridge 'gnss' init ok
uavcan: sensor bridge 'mag' init ok
uavcan: sensor bridge 'baro' init ok
nsh> uavcan status
Pool allocator status:
        Capacity hard/soft: 500/250 blocks
        Reserved:  2 blocks
        Allocated: 2 blocks
UAVCAN node status:
        Internal failures: 0
        Transfer errors:   0
        RX transfers:      0
        TX transfers:      55
CAN1 status:
        HW errors: 27
        IO errors: 27
        RX frames: 21
        TX frames: 37
CAN2 status:
        HW errors: 4300
        IO errors: 4300
        RX frames: 0
        TX frames: 37
ESC actuators control groups: sub: 0 / req: 0 / fds: 2
ESC mixer: NONE
Sensor 'gnss':
RX errors: 0, receiver node id: N/A

Sensor 'mag':
devname: /dev/mag
channel 0: empty
channel 1: empty
channel 2: empty
channel 3: empty
channel 4: empty

Sensor 'baro':
devname: /dev/baro
channel 0: empty
channel 1: empty
channel 2: empty
channel 3: empty
channel 4: empty

nsh>
```
however I cannot get connect via MP terminal as terminal support was removed from the firmware. Thanks.

benjammin1988, 2017-02-04:
Did you verify external mag is being used with a piece of metal?
Ben

davidbitton, 2017-02-04:
No I haven't. I was _assume_ I should see multiple compass values in the Status tab of MP. Let me check.

davidbitton, 2017-02-04:
A live compass calibration failed for compass #1 which I'm guessing is on the Zubax.

benjammin1988, 2017-02-04:
The problem I have is everything appears to check out (even calibration is successful).  But If you are to manually check the compass with metal, you'll see that the internal compass is the only one actually being used by PH2.

WickedShell, 2017-04-21:
@EShamaev this appears to still be an issue. Would you have a chance to look at this? (Additionally are you tracking detecting a GPS being detected after the system has been booted? Or is that just not on the agenda for UAVCAN gps's?)

EShamaev, 2017-06-08:
This issue and last message is date far before new CAN drivers.
So I will safely close this one, however we can continue on https://github.com/ArduPilot/ardupilot/issues/6400

closed 2017-06-08 by EShamaev