ArduPilot/ardupilot #5275: MAVlink library prevents OSD from receiving RSSI from 3DR modems

night-ghost, 2016-11-25:
#### Issue details
file https://github.com/ArduPilot/ardupilot/blob/master/libraries/GCS_MAVLink/MAVLink_routing.cpp

lines starting 103

```
if (msg->msgid == MAVLINK_MSG_ID_RADIO ||
        msg->msgid == MAVLINK_MSG_ID_RADIO_STATUS) {
        // don't forward RADIO packets
        return true;
    }
```

this code fully filters out MAVLINK_MSG_ID_RADIO_STATUS and MAVLINK_MSG_ID_RADIO messages so no one module (and OSD too) can't see RSSI from 3DR modems.

So in Copter 3.1 3DR RSSI is working, and definitely not working on Copter 3.3/3.4. All OSD users logically [consider](https://www.rcgroups.com/forums/showpost.php?p=36273947&postcount=1861) that it is problem of OSD itself

This messages should be filtered out only on port from which they are comes, and be routed to all another MAVlink ports. Or even better to route this message to all ports and filter them out in 3DR modem firmware.

imrj, 2016-11-27:
can someone please reply to night-ghost? this is a long standing issue we would like to fix in MinimOSD

OXINARF, 2016-11-28:
@imrj I'm not sure what you want us to answer. This filtering was certainly done on purpose but I can see why other components would like to have that information. For those reasons I marked this as an enhancement.
As always PRs are welcomed, although I can't say if it would be accepted or not. In dev calls we usually discuss concrete opened PRs or an issue someone brings to know if he should proceed with implementation. So if someone wishes to implement this there are two possible courses of action:
- go to a dev call and ask to discuss this issue to see if the team approves it or not
- implement it and do a PR and then wait for it to be merged or rejected

night-ghost, 2016-12-06:
As for me, this is the bug because that PR which brought these changes was not sufficiently tested

guiseco, 2017-03-31:
what about this issue ?

tizianofiorenzani, 2018-03-16:
I totally find this issue pretty important. As for now my companion PC cannot read the connection status.

tridge, 2018-03-19:
we've decided to add a SERIAL_OPTIONS bitmask parameter with a bit to change this behaviour