ArduPilot/ardupilot #5281: mode change by MAVLink

credtiger96, 2016-11-26:
#### Issue details
Ardupilot code use 
MAVLink uses mavlink message SET_MODE (#11) but according to mavlink document, 
it is deprecated and recommended to use COMMAND_LONG with MAV_CMD_DO_SET_MODE.
but ardupilot seems that dose not support that interface. 

#### Version
_What version was the issue encountered with_
latest version

#### Platform
[  ] All
[  ] AntennaTracker
[  o ] Copter
[  ] Plane
[  ] Rover

OXINARF, 2016-11-26:
It looks like only Copter doesn't support it (although support in the other vehicles could probably be better). PRs are welcomed.

peterbarker, 2017-12-19:
I believe all vehicles now support this.

closed 2017-12-19 by peterbarker

josezy, 2020-03-27:
> I believe all vehicles now support this.

since which version for ArduCopter?