ArduPilot/ardupilot #5282: SUMD connection not robust

Benoit3, 2016-11-26:
#### Issue details
To improve robustness of SUMD interface with graupner receivers, I noticed that I need to increase 
the size of the "static uint8_t dsm_buf" buffer to 64 instead of 32.

I do not understand why, as controls_tick() seems to be called each few milliseconds (necessary to measure the inter-frame delay).

I'm ready for further investigations, but I don't find some informations : 

- I know how to connect to FMU console, but not  to the px4io one. Where are  RX/TX pins for px4io console???
- what is the standard right for controls_tick() calls?
- what's the size of serial rx buffer of uart driver ?

Regards,
Benoit

#### Version
Arduplane 3.7.1

#### Platform
[ X ] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover

#### Airframe type
_What type of airframe (flying wing, glider, hex, Y6, octa etc)_

#### Hardware type
_What autopilot hardware was used? (pixhawk, pixracer, PX4FMU etc)_

#### Logs
_Please provide a link to any relevant logs that show the issue_

Benoit3, 2016-12-22:
After investigation it appears that with some modification it is possible to improve robustness of SUMD decoding up to 100%. These modifications are :
- use the the servo rail to power the receiver, not the RC in power supply. As now receiver implement a transmitter (for telemetry), power consumption show some peaks. The RC in power supply seems to be done through a output pin of px4io processor, not always strong enough to power a GR16 receiver : if you connect GR16 after pixhawk power up, the PX4IO processor is crashed 100% of the time

- connect ppm input and sbus input to the ground. I observed some false detection on one of these input with an average of one time every  half hour. To avoid them I used two solution : connect inputs to the ground and deactivate ppm and sumd decoding when some signal is detected on dsm input

- deactivate dsm decoding (not protected by checksum to verify integrity of received serial data) when serial stream satisfy st24, sumd or srxl checksum criteria

I also improve px4io driver to ease rc input monitoring which was not up to date for sumd protocol

I checked these modification in flight on an Arduplane 3.7.1 software and on bench after merge with Arduplane 3.8 beta.

Not that it is possible now to connect to receiver on the pixhawk with priority in the following order : -st24/sumd/srxl/dsm
-sbus
-ppm

I will propose a pull request in following hours to close this issue.
Benoit

peterbarker, 2021-07-30:
We've got completely fresh implementations of these RC protocols how, with a completely different (possibly overlapping!) set of bugs.

Thanks for working on this, 'though

closed 2021-07-30 by peterbarker