ArduPilot/ardupilot #5286: Unable to disarm after landing with EKF/Compass errors.

Gregsky, 2016-11-27:
#### Issue details
The Aircraft reported compass and EKF errors during the flight, after landing it would not disarm with the radio or the Tower app, the Tower app was still showing the land button, the EKF diagnostics ware showing high magnetic field and it looked like the landing was not detected.
I have stopped the motors with the on-board safety switch.
Should the EKF/compass error stop from disarming?

#### Version
3.4.2
#### Platform
[  ] All
[  ] AntennaTracker
[ x ] Copter
[  ] Plane
[  ] Rover

#### Airframe type
Quad X
#### Hardware type
dropix v2 (pixhawk compatibile)
https://drotek.com/shop/en/home/478-dropix-flight-controller.html

#### Logs
https://www.dropbox.com/sh/xdnj4smv6yfbdh6/AADrcxfap75t9NVAPNgbzzRZa?dl=0

Gregsky, 2016-11-27:
I think this log (2016-11-27 14-53-12.log) shows the issue, you can see three failed attempts to disarm after landing.
Other logs are post compass calibration.

IamPete1, 2021-08-15:
sorry we never got to this.

closed 2021-08-15 by IamPete1