ArduPilot/ardupilot #5287: GPS values need to be increased from 7 decimal places

mboland, 2016-11-27:
#### Issue details
Enhancement Request
Increase GPS internal accuracy from 7 decimal places to at least 9

For any sub cm system more decimal places are required.
7 decimal places = 1.5cm accuracy at best
With the range of  RTK systems on the rise claiming sub cm accuracy Ardupilot is **not** ready for any of these.

#### Version
All - Currently 3.4.2

#### Platform
[X] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover

#### Airframe type
All - Primary interest Copter, then Plane, but all versions would be improved with better accuracy.

#### Hardware type
All - Identified on Pixhawk, Pixhawk2.1, soon to be using NavIO+

#### Logs
http://uavs.com.au/Arducopter/logs/2016-11-11

magicrub, 2016-11-27:
This is done using integer lat long instead of float. There is a MAVLink cmd long that handles this.

WickedShell, 2016-11-27:
@magicrub command_int, mission_int doesn't provide enough places. Command_long is a float storage, and internally we have to actually store/work on a higher precision to support this.

magicrub, 2016-11-27:
... oh

gmorph, 2016-11-28:
Hi, I'd like to understand this further.  This page was illuminating:
https://gis.stackexchange.com/questions/8650/measuring-accuracy-of-latitude-and-longitude
To summarize:
Earth's equatorial circumference is about 40,000 kilometers.
A latitude/longitude value breaks that distance up into 360 degrees, starting at -180 and ending at 180.
This means that one degree is 40,000 km divided by 360:

    40,000 / 360 = 111

So, one degree is 111 kilometres.  For fractions of a degree, you divide it by 10 for each decimal place

   decimal
   places   degrees        distance
   -------     -------            --------  
   0           1                   111   km
   1           0.1                11.1  km
   2           0.01              1.11  km
   3           0.001            111   m
   4           0.0001          11.1  m
   5           0.00001        1.11  m
   6           0.000001      0.111 m
   7           0.0000001    1.11  cm
   8           0.00000001  1.11  mm

The COMMAND_INT supports the int32_t for the lat, long which represents a range from from −2,147,483,648 to 2,147,483,647.  So if I had a mm accurate co-ordinate like
179.00000001 I would represent it like 17900000001 which is 1 "digit" longer then 2147483647
then that can't be represented by a int32_t.  So we can be cm accurate as the in32_t has enough digits for that but not mm accurate?

On a separate note not sure there is an immediate need to be sub 1.5cm accurate in ArduPilot.  Our vehicles aren't exactly capable of that small an adjustment in location.  If you had a copter hovering and need to go left 1.5cm I think that might prove tricky.

Thanks, Grant.

WickedShell, 2016-11-28:
@gmorph the normal target would be 9 decimal places (this is at least the accuracy the u-blox high accuracy supports IIRC).

I'd actually agree that this is a nice long term goal but not an immediate requirement. The reason @mboland is having problems is that he's looking to use the APM log for camera tagging, where we are having accuracy limitations from within the log for the camera stuff. In his particular case the SBF board he is using is providing all the required logging to the needed accuracy, but for using Trimble/Applanix/U-Blox there isn't an equivelent on board logging. The first step that would be helpful would be just to log higher accuracy for camera tagging purposes (which is a smaller change set then supporting that high an accuracy in flight).

mboland, 2016-11-28:
We do only use the Septentrio logs, due to this lack of decimal accuracy in Ardupilot, but unfortunately, the way RTK is done on Ardupilot is by injecting the RTK signal through Mission Planner to Ardupilot which then routes it to the GPS.
And this is where we lose our decimal accuracy, as the RTK corrections coming in are truncated to 1e7.
As Ardupilot, and hence Mission Planner, use only 1e7 then no matter how good our RTK is, it is truncated by the transmission to our GPS board.
Even if this were improved to at least 1e9 it would greatly improve things.
It would have to be a 2 part fix, Ardupilot AND Mission Planner.

meee1, 2016-11-28:
I'm just going to clear a few things up here.
the Septentrio logs internally at the full rtk precision (double). The data Mission Planner passed from the ground to the gps is also passed in full precision. this is the flow

base gps outputs rtcm > mp inject gps reads rtcm > mp passes the full rtcm message unmodified out to the telemetry radio > the uav receives this, and outputs it to the gps, again unmodified in any form. 
This data contains the positon of the base station, and the pseudorange and carrier phase information. All of this is generated by the base gps.

Where loss is expected is, in the uav gps driver, because of a convert from double to 1e7 (ie 7 decimal places).
In the EKF, because internally it is using floats. ~7.2 digits. All ekf measurements are based around an origin (1e7), so the closer to the origin the more accurate it is.
Camera Trigger logging, because this is derived from the fusion position of the ekf. not the raw gps sample that could be up to 200+ms out.
DF logging, depending on the message as it will have different accuracy's. ie GPS will be 1e7, POS will be ~7.2 digits, as its derived from the ekf.

To improve the accuracy as it stands at the moment, Doubles would need to be used internally.

so in a nutshell, the rtk gps onboard logging will be at full precision.
ArduPilot precision will be at best 1e7.

mboland, 2016-11-28:
Thank you so much @meee1 for clearing that up.
This we can work with.
I do think that the longer term goal should be to increase precision in Ardupilot, but at present Septentrio do their own logging with camera trigger detection and it will do the job nicely without the need for Ardupilot logs.

magicrub, 2016-11-28:
As an incremental step, the GPS driver should probably start converting the internal storage variables to doubles. Some of the maths are already double precision, just need to do more of that and stir those results without losing the precision. The internal maths are currently mostly the 32bit integer flavor. I think Floats are mostly just in the comm links.

Are some point soon well need a new double precision MAVLink packet. It is, obviously, an inevitable progression.

amilcarlucas, 2018-06-19:
So with 7 digits precision and no vibration, is it to expect that the EKF position innovations would be at most 1.11 cm?

jimovonz, 2020-07-18:
I have implemented some code to address this issue. It adds two additional attributes to the Location object that represent an additional 2 digits of precision for latitude and longitude. All the location calcs have been updated appropriately. There is support for the UBX NAV-HPPOSLLH message to get the added precision from modules that support it. I have also added a mechanism for uploading WP's with the added precision and dataflash logging.

https://discuss.ardupilot.org/t/chasing-better-accuracy/58966