ArduPilot/ardupilot #5289: Rover: RCx_REV does not reverse manual or steering RC input - BUG

mikerob, 2016-11-27:
#### Issue details
The RCx_REV setting does not reverse the direct RC input, it seemingly only reverses the flight controller's input. 

I encountered it in steering mode.  The vehicle was veering off to the side - the RC output would track the steering input in general, but with an offset.  If I steered to the right, the rover would keep going to the right, and same for left.

Reversing the channel (RCx_REV) had no effect in MANUAL mode (didn't reverse the steering), and did not change the overall steering direction, (stick left still turned left) but did fix the problem (it now tracked straight).

It looks like the direct RC portion of the input is not getting reversed. Is this intended behaviour? 

#### Version

3.0.x latest Beta (25 Nov)

#### Platform
[  ] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[ o] Rover

#### Airframe type
1/10 scale electric RC car

#### Hardware type
Chinese Pixhawk 2.4.8

#### Logs
Can generate a log if required.

gmorph, 2016-11-28:
If I understand what you've written then the Rover is behaving correctly.  Manual mode is just that - fully Manual.  In Rover (unless your using skid steering) if you put it into Manual the inputs to the AutoPilot coming from your RC Transmitter will be passed straight through to the output - no ArduPilot adjustments.  It doesn't look at the RCx_REV parameters.
However, Steering mode is an ArduPilot control mode so it does look at the RCx_REV parameters.  So, when your in manual mode if you need to set channel reverse on your transmitter for your Rover to steer in the correct direction then you need to set that RCx_REV so the AutoPilot knows to reverse it when its doing the driving and not you via the transmitter.  Make sense?  When in Steering mode ArduPilot is doing the actual driving (its not straight pass through) and its just taking input from the transmitter which it does calculations on. Its there mostly so people get throttle and steering directions correct.
Let me know how you go.

Thanks, Grant.

mikerob, 2016-12-01:
OK, I understand what you're saying. I think the confusion was as a result of Mission Planner - when you reverse a channel in MP, the green bars go in the opposite direction AND the indicated PWM (on the RC config page) switches direction - what was 1100 for left stick becomes 1900 for left stick once reversed.

These values are NOT however, reversed when they get to the autopilot - RC IN would still show 1100 in both cases above.

closed 2016-12-01 by magicrub