ArduPilot/ardupilot #5290: Can I use a radio signal (3DR 433MHZ) as input to the SITL?

JoseBelz, 2016-11-28:
#### Issue details

Hi,
I want to study the mavlink protocol link via radio, using 3DR 433/900 MHz radio kit.
I have thought about using the SITL simulator to capture frames and to study the link more easily without having to use a real drone.

I know how to use the SITL via USP and TCP but I do not know if it is possible to use as a command input a serial port. 

I've been looking but I can not find anything. Could you help me?

thanks in advance

squilter, 2016-11-28:
Maybe this conversation will help? ArduPilot/ardupilot_wiki#419

OXINARF, 2016-11-28:
@JoseBelz Yes, you can connect a real radio to SITL. You'll need to pass an UART option when you execute SITL with the following format: `uart:/dev/ttyUSB0:57600`. The path is the path to your radio device and the number is the baudrate and is optional (default will be used if not provided).

Now, this is a development support question and those should be posted at the forum (http://discuss.ardupilot.org) or asked about in Gitter (http://gitter.im/ArduPilot/ardupilot).

closed 2016-11-28 by OXINARF