ArduPilot/ardupilot #5291: Guided Mode msg_set_position_target_global_int not using WPNAV_SPEED

billbonney, 2016-11-28:
#### Issue details
Guided Mode using smsg_set_position_target_global_int with location only does not obey the WPNAV_SPEED setting, (or any speed setting). while maxes out at about 3.8m.s

Using msg_set_position_target_global_int with loc&velocity settings causes the vehicle to crash :(

#### Version
AC3.4.2

#### Platform
[  ] All
[  ] AntennaTracker
[ x ] Copter
[  ] Plane
[  ] Rover

#### Airframe type
SITL and Iris

#### Hardware type
AWS(SITL) and PIxhawk1

#### Logs
I post some logs later today, but it's easy to replicate, just seta Guided Point in Mission Planner, APM Planner 2.0 Tower, MAV Pilot etc..

billbonney, 2016-11-29:
@rmackay9 @tridge 
I tried trigger of the leash every iteration in AC_WPNav::update_wpnav() https://github.com/ardupilot/ardupilot/blob/master/libraries/AC_WPNav/AC_WPNav.cpp#L751

but no joy, I'm still stuck at 3.8m/s top speed. I verified that PosControl and AC_WPNav are both being set to 1000 (10m/s). The codes really new to me, so I'm not really sure how to fix it.

Any help appreciated.

billbonney, 2016-11-29:
Why isn't this marked as a bug? 

Isn't there anybody who can make a suggestion to send guided waypoints to enable speeds greater than 3.8m/s?

magicrub, 2016-11-29:
Can you post your param file?

billbonney, 2016-11-29:
WPNAV_SPEED is set to 1000

I've added printf to the code and I can see that it's being set to 1000 and 1100 from mission and parameters. It's being passed into the position controller.

[parameter.txt](https://github.com/ArduPilot/ardupilot/files/619299/parameter.txt)

[kits_speed_mission.txt](https://github.com/ArduPilot/ardupilot/files/619310/kits_speed_mission.txt)

khancyr, 2016-11-29:
hi,
The bug is old. (see https://github.com/ArduPilot/ardupilot/issues/3536) . From what I remember , it seems that pos controler has priority over speed , so it set speed according to pos controler. But if you use extrem value for velocity, it will use only velocity ... I don't if it could help to find the issue

billbonney, 2016-11-30:
I have just tested this against AC3.3.3 and it works as expected with it achieving the desired, in this case, 8m/s set in the WPNAV_SPEED parameter.

rmackay9, 2016-12-05:
I've checked this with master, and although it can't quite reach it's max speed it does get close.  It gets up to 8.4m/s instead of the desired 10m/s.  WPNAV_SPEED was set to 1000. 
[image]

rmackay9, 2017-02-15:
I think Bill said he was re-testing or couldn't recreate the issue or something like that so I'm closing this.  We can reopen if anyone complains.

closed 2017-02-15 by rmackay9