ArduPilot/ardupilot #5292: Plane: increase target airspeed on AUTo land via head wind

magicrub, 2016-11-28:
Add half of dot-product of head-wind to airspeed.

WickedShell, 2016-11-29:
Why would you increase it by this factor? Unless you are not maintaining any forward ground speed I'm struggling to see why this would be a good thing without more explanation.

magicrub, 2016-11-29:
Tridge had a recent minor crash where large wind gusts caused a loss of airspeed just before landing. The larger the head wind the larger the gusts. Also, that will imply a slower ground speed so there shouldn't be a problem to bump up airspeed a little in that case.

WickedShell, 2016-11-29:
The problem I could see with bumping the airspeed is that you could be generating far to much lift to match the descent profile that you are looking for, which to balance the airspeed can lead to a nose down attitude. By raising the ground speed you are also effectively raising the desired vertical descent rate which can lead to higher airframe and landing gear stresses during landing.

I can see the overall desire for this but I think it requires a lot of care, it would be very easy to end up with  a system of competing desires that leads to instability/more damage. (I'm thinking of flights I've done with winds that are higher then my stall speed, if you were to add that amount of wind to my normal landing speed you would be looking at flying at far to high an airspeed to safely belly land the aircraft).

magicrub, 2016-11-29:
If you have a strong head-wind then you have a low ground speed and are
covering a much longer and shallower glide slope. The flare is a vertical
velovity rate, not air/gndspeed based.

Remember, your airspeed is irrelevant when measuring ground impact, it's
all about ground speed. This effect will never increase your ground speed
to be higher than when landing with no wind. Also keep in mind this is the
component of the head-wind. A cross-wind would contribute no airspeed
increase.

WickedShell, 2016-11-29:
What I'm worried about is the nose down attitude that I could see this easily causing.

magicrub, 2016-11-29:
Are you saying it would throttle-up and nose-down? It will still follow the
glide slope though, I don't understand how that causes a problem. On
another note, the pre-flare could be treated special and override this
"feature".

magicrub, 2016-12-03:
fixed via https://github.com/ArduPilot/ardupilot/pull/5305

closed 2016-12-03 by magicrub