ArduPilot/ardupilot #5294: running self-wrote library to cause safety switch disarm and gcs lose connection

CrazyPopLin, 2016-11-29:
#### Issue details
I am useing the master version of arducopter. I wrote a library to get serial data from uartE port and setup 115200 in **SerialManager.cpp**. This library is running at 400hz loop and invokes by other control loop. The safety switch is on from start up.     
Mostly, it works fine. But sometimes, the safety switch will automatically off (like you can hear ESC sound) and mission planner will lose connection, and the B/E LED in IO part will orange flash (or red flash). The test in RTL mode which not call my library.  
Any one can give me some suggestions? thanks so much

#### Version
Master version of ArduCopter
#### Platform
[  ] Copter

#### Airframe type
copter

#### Hardware type
pixhawk

#### Logs
No logs

magicrub, 2016-11-29:
Do you have a link to your code? The official code does not do that so your changes are the cause. We don't know what your changes are.

CrazyPopLin, 2016-11-30:
[my ugly code](https://drive.google.com/drive/folders/0B8u9tW-Q7BBmMTF1T2dfUlRCUmM?usp=sharing), thanks for your help.

CrazyPopLin, 2016-11-30:
Serial.write('<');
      Serial.write(a >> 8);
      Serial.write(a & 0xFF);
      Serial.write(d >>8);
      Serial.write(d & 0xFF);
      Serial.write('>');

The serial data form like this, first byte is start_flag, the two and three bytes are angle, four and five bytes are distance, last byte is end_flag.

magicrub, 2016-11-30:
Do you have a git branch we can see instead of random loose files? We can't see how that interacts with the codebase.

CrazyPopLin, 2016-11-30:
hi @magicrub , I upload the modified branch code in my [branch](https://github.com/CrazyPopLin/ardupoilot/blob/master/libraries/AP_Tmxk/AP_Tmxk_LIDARScanner.cpp). if I uncomment **line 94** which is `process()`, the code works fine in 20 tests. And if not, the pixhawk will dead in 5 tests. Please give me some suggestions and code review if it's possible. And forgive my ugly code. thanks.

CrazyPopLin, 2016-12-01:
After tested, it's problem of `doProcess()`. the problem still there.

CrazyPopLin, 2016-12-06:
I've fixed the issues. thx for your help.

closed 2016-12-06 by CrazyPopLin

rmackay9, 2016-12-06:
Excellent, great news.  Txs for the follow-up.