ArduPilot/ardupilot #5300: Feature request - Support for Dshot ESC protocol

dk7xe, 2016-11-30:
#### Issue details
Dshot is a new "digital" ESC communication protocol currently developed by flyduino (Felix Niessen) together with borisb (betaflight). BLheli and KISS ESC are supporting this new protocol already.
-> https://www.rcgroups.com/forums/showthread.php?t=2756129

Advantages (as described by Felix on rcgroups) compared to the most used analog signals like PWM as also oneshot125 or 42 are:
- no signal jitter.. if the FC sends 1375 the ESC will receive 1375
- high resolution (2048steps)
- no oscillator drift (nomore calibrateing ESC's)
- more robust against spikes
- safer as every singnal has a CRC (cyclic redundancy check)
- maybe later some goodies like having the ESC's beep with the buzzer signal or things like that.

Since arducopter aims to support smaller/racer frames as well i suggest to add support for Dshot.

#### Platform
[  ] All
[  ] AntennaTracker
[X] Copter
[  ] Plane
[  ] Rover

#### Airframe type
all

#### Hardware type
Pixhawk

magicrub, 2016-11-30:
Would be so much nicer of they would just use CAN and get on with it..

R-Lefebvre, 2016-12-01:
Ooh, don't go there. They laughed at me last time I suggested it. Too complicated and expensive.

proficnc, 2016-12-01:
CAN .... I went there :)

lthall, 2016-12-01:
This is an important step forward and we should make sure this is supported in the next release.

While CAN ESC's will be great. BL_Heli represents mainstream ESC state of the art. It is benefiting from many thousands of racing quads that need to push the control capability to it's limit. 

This is another fantastic step forward for mainstream ESC development and we should support them not only because it is beneficial to us but also because they are a fellow open source community that is doing great things.

lvale, 2016-12-01:
Second @lthall words. 

While CAN has the "potential" to be all for all, the community moves forward with easier/cheaper and effective solutions.

tridge, 2016-12-01:
I think we should try and support this. The critical component we need is "DMA to a pin". I'm also not sure if we have enough DMA channels available. We may have to have options to choose between some of the other uses of DMA (eg. UART RX and SPI) and this.
It is likely that this will be a fairly tricky feature to develop, but I also think it is very worthwhile. The same basic system level features are needed to support FW update of ESC by flight controller.

magicrub, 2016-12-01:
are the GPIOs memory mapped? The PIC32 on using MIPS arch has that and it
is quite handy. You can have DMA stream any thing via any event timing or
source to any GPIO pin because they're all memory mapped.

dk7xe, 2016-12-07:
@tridge will u include support for the telemetry feedback from the ESC as well? This will eliminate the need for e.g. a current sensor because data can be read from ESC (if they have one like the KISS24re ESC).

auturgy, 2016-12-22:
@tridge if you're going to do this I can donate some dshot esc's.

dk7xe, 2017-01-05:
I can donate KISS24RE ESC as well if needed.

magicrub, 2017-01-05:
I think this work may be underway.

iskymaster, 2017-06-01:
Can someone please explain me what CAN is?

auturgy, 2017-06-01:
https://en.m.wikipedia.org/wiki/CAN_bus

Regards,

James

iskymaster, 2017-06-01:
James, Thanks!

May be the reason people have not jumped on it is because of the royalty fees?

**"Bosch holds patents on the technology, and manufacturers of CAN-compatible microprocessors pay license fees to Bosch, which are normally passed on to the customer in the price of the chip. Manufacturers of products with custom ASICs or FPGAs containing CAN-compatible modules need to pay a fee for the CAN Protocol License."**

cmanley, 2017-08-17:
Is there any news or progress on this? Just about every new ESC sold these days uses Dshot, so it would be a pity to not be able to use them and also not have to calibrate ESCs anymore.

netptl39, 2017-09-11:
Hi All, 
Is there any news on this issue?
I can also see the advantages of CAN vs Dshot, but: CAN ESCs are 80-100USD/pc while  32bit DShot1200 capable ESCs are 25USD/pc tops. BLheli_32 is out for awhile now and have ESC telemetry too. 
I am not a racing at all, but I could sure use less jitter on the copters.

Saxin, 2017-10-19:
I don't recall ardupilot motor update rate, but i assume its on the order of 1k or so...
this makes dshot control rate gains irrelevant, yet the telemetry side of things, erpm and current can give some edge if used properly.
esc calibration will be absolute - a bonus.

xkam1x, 2018-01-21:
I initially wanted to post comment here but ended up in PX4 issue on Dshot. Original comment:
I have been reading several forums about ESC protocol and it seems that digital bi-directional communication is a way to go.

The options are then i2c, CAN, UART and Dshot.
with i2c we have 400kbit/s.
CAN 1Mbit/s.
UART 1Mbit/s (depends on hardware).
Dshot 600kbit/s,

Since CAN needs licensing, it is probably best to avoid; otherwise cost of ESC will be higher and its adaptation will be slow.
i2c maybe a possible solution but I fear data integrity as SCL and SDA need to be synced.
UART is probably best as it does not require licensing and generally found on all micro controllers.
Dshot uses DMA which maybe difficult to program.

Personally I would like to see UART based ESC's and this how it can be done:
Hardware:

Invert UART signals.
Connect TX and RX togather.
Use 2 data lines, one with default pull high and other with default pull low to get differential signals (requires extra hardware but better reliability).
Share the UART bus with all ESC's.
Data packets could be sent like:
Sync1,Sync2,Motor 1 MSB,Motor 1 LSB,Motor 2 MSB,Motor 2 LSB,Motor 3 MSB,Motor 3 LSB,Motor 4 MSB,Motor 4 LSB,Data Request,Checksum

If whole message checksum has more overhead then we could use 11 bits for throttle value and 5 bits for checksum giving us 2048 levels and still keeping within 2 bytes.

Data request could contain first 4 bits as ESC ID and last 4 bits for data request. This allows for 15 motors and 15 different data packet structs, assuming last byte 0x00 means response not requested and it could be used to perform load balancing.

As soon as one message is sent, one of the ESC would respond with data packet requested and FC would process before requesting another.

This type of system does require ESC ID's to be set before hand but it is a onetime requirement.
Also we could change the Sync1 and Sync2 to indicate that the packet is throttle values or config parameters.

What do you think?

auturgy, 2018-03-12:
Update: WIP still, but not far away from flight testing 
https://github.com/tridge/ardupilot/tree/chibios-dshot-wip

demondev, 2018-03-15:
Can I ask if telemetry feedback from the ESC is being implemented also ? I to would be very happy to get rid of the current sensor, and gain voltage, current, RPM, and ESC temperature feedback.

tridge, 2018-03-22:
@demondev yes, I am going to implement the telemetry feedback.
Status so far is:

- dshot works (150, 300, 600 and 1200)
- all done with DMAR so uses minimal number of DMA channels
- all auto-configured using hwdef.dat for the board
- we've had several successful test flights
- I have successfully done pass-thru configuration of both BLHeli_32 and BLHeli_S ESCs using the USB port on an ArduPilot board, using the BLHeliSuite programs on windows
- I've successfully upgraded firmware on ESCs using ArduPilot passthrough

I will be looking for people to do testing over the next few days. Please join the https://gitter.im/ArduPilot/ChibiOS gitter channel to join in the fun.

I'd also like to support Kiss ESCs, but haven't bought one yet. Donations welcome :-)

dk7xe, 2018-03-22:
@tridge i have send a contact request via skype a few minutes ago. Maybe i can help u on KISS ESC.

tridge, 2018-03-24:
@dk7xe got it just now (for some reason skype on linux didn't show it, but skype on my windows box did)

dk7xe, 2018-03-26:
@tridge 4 KISS32A ESC are ordered now. Hope they will arrive soon at your end. Thanx for your work!

padvinder95, 2018-04-04:
Have mentioned something in gitter but should maybe add here too: I am really looking forward to using this in Plane, too, since I use Aikon AK32 BLHeli32 ESCs and would love to use telemetry (and d-shot is nice too). Is there anything I can do to help, such as send some of those ESCs for testing, test something here myself, etc.?

massimiliano-mantione, 2018-04-06:
Is there any specific reason why Dshot is currently not supported on rovers?

I am starting a new project, it is a small autonomous ground vehicle for various speed competitions (line following, track races with other vehicles, maze solving...).

I am still at the "hardware selection" phase, and having fast and accurate wheel speed control (with RPM telemetry) would be really nice :-)
Is it such a crazy idea to use BLHeli32 ESCs to accurately drive wheels instead of propellers?
(maybe it is, but I'd like to know...)

magicrub, 2018-04-06:
You will want to use an encoder for that

cs8425, 2018-04-06:
@massimiliano-mantione 
I think that's possible, but not so accurate as you thought.

rmackay9, 2018-04-07:
This is in master so closing.  hope that's OK.
I don't know why BLHeli and Dshot aren't supported on Rover but I suspect it's easy to add.

closed 2018-04-07 by rmackay9

massimiliano-mantione, 2018-04-07:
Thanks for the replies, I'll look into using an encoder!

awright424, 2018-04-11:
any idea of when this  will be available on arm boars? specifically bbbmini/pocketpilot?