ArduPilot/ardupilot #5301: Joystick Trim Problem in Ardupilot:Plane for Quadplane setup

ferdicakmak, 2016-11-30:
Hello,

When I connect and setup my joystick it works fine with Mission Planner 1.3.41.8. When the controls on joystick is centered, the outputs shown with green bars (see below figure) are centered and when I move them the outputs move accordingly.

[image]

When I connect to the aircraft (ArduPlane V3.7.1) via telemetry, I see that the outputs of the joystick are set to minimum, therefore trying to move my gimbal to minimum position. When I check the configuration parameters I see that RC5_TRIM, RC6_TRIM and RC5_TRIM are set to 1104 (minimum position), which causes this problem. 

[image]
[image]
[image]

When I set RC5_TRIM, RC6_TRIM and RC5_TRIM to 1515 (mid position) everything works perfectly and joystick outputs are centered and gimbal moves as expected with joystick.

[image]
[image]
[image]

The problem;
Since I am using ArduPlane in QuadPlane mode (Q_ENABLE = 1) with VTOL ESCs connected to RC_OUT5-8, I believe that RCX_TRIM values are updated according to Q_* parameters and set to minimum, everytime autopilot restarts. Which makes me set RC5_TRIM, RC6_TRIM and RC5_TRIM to 1515 manually again before every flight.

Solution Proposals:
1.	RCX_TRIM values may not be updated with Q_* parameters if not necessary for flight in Q_* modes.
2.	RCIN and RCOUT settings (MIN, MAX, TRIM) may be separated so that they do not effect each other.
Is it possible to fix this problem in ArduPilot:Plane codebase? Or coluld you reccommend other solutions?

Thanks,
Ferit.

#### Version
Arduplane V3.7.1

#### Platform
[  ] All
[  ] AntennaTracker
[  ] Copter
[X ] Plane
[  ] Rover

#### Airframe type
Quadplane

#### Hardware type
Pixhawk

#### Logs
-

ferdicakmak, 2016-12-01:
Hello,

Is there anyone who can tell me if RC5-8_TRIM values are utilized for QuadPlane flight for Q modes?

I am trying to **decide if it is safe to manually change RC5-8_TRIM values**, after QuadPlane V3.7.1. boots up and overwrites RCX values according to Q_* parameters since Q_enebled.

Any idea?

Thanks,
Ferit.

magicrub, 2017-01-14:
I don't know quadplane very well and have never used a gimbal so I've never tested the interaction. Could this be related to param [RC_TRIM_AT_START](http://ardupilot.org/plane/docs/parameters.html#rc-trim-at-start-rc-trims-auto-set-at-start)?

IamPete1, 2021-08-15:
I think this has been fixed, I don't see any code path where we change RC trims from Quadplane. The `RC_TRIM_AT_START` param is also gone.

closed 2021-08-15 by IamPete1