ArduPilot/ardupilot #5302: ArduPlane: QuadPlane, ATT.DesYaw is not logged

ferdicakmak, 2016-11-30:
#### Issue details
Hello,

I am using Pixhawk with ArduPlane V3.7.1 for QuadPlane setup. I see that ATT.DesYaw is logged as 0 in dataflash along the whole flight, when in QSTABILIZE, QHOVER and FBWA (other modes are not tested). Other variables of ATT like DesRoll, Roll, DesPitch and Pitch are fine.

[image]

Any suggestions or solutions?

Thanks,
Ferit.

#### Version
ArduPlane V3.7.1

#### Platform
[  ] All
[  ] AntennaTracker
[  ] Copter
[X] Plane
[  ] Rover

#### Airframe type
QuadPlane

#### Hardware type
Pixhawk

#### Logs
-

expntly, 2017-04-18:
@magicrub @tridge Any idea why this is not being logged?  Would be helpful for yaw tuning.  Happy to try putting together a PR if the change doesn't require much knowledge.

expntly, 2017-07-01:
@peterbarker maybe knows?

ferdicakmak, 2017-07-01:
I think it is fixed in ArduPlane 3.8.0beta5. ATT.DesYaw is logged correctly.

peterbarker, 2017-07-03:
Commit fedabd1ace23797d6eeb75b351f7d72029e93881 added this logging.

closed 2017-07-03 by peterbarker

expntly, 2017-07-15:
Just loaded 3.8.0-beta6.  Is this really working?

[image]

expntly, 2017-07-16:
Here's another flight and its log.  Logging isn't empty at least like before but I'm seeing strange spikes.

[image]

DF log: https://www.dropbox.com/s/v644adu7iu7lss8/00000028.BIN?dl=0

OXINARF, 2017-07-17:
@expntly I don't see anything strange. Yaw range is [0; 360[ so when it is around 0/360 it will show those "spikes" in the log.

expntly, 2017-07-18:
@OXINARF I see.  I guess I was also surprised by DesYaw and Yaw not following one another like Roll/DesRoll and Pitch/DesPitch would.  DesYaw is so flat that I'm suspecting something is wrong and this is not related to tuning?

expntly, 2017-11-11:
@peterbarker Looking at a log file I have with a transition from VTOL to cruise, ATT.DesYaw goes to 0 after the transition to cruise.  Is that expected?  Roll and Pitch are logged just fine in both modes.

See below at around 3.35, then it comes back at around 3.44 when de-transitioning.

[image]

ArduPlane V3.8.2.

peterbarker, 2017-11-13:
On Sat, 11 Nov 2017, expntly wrote:

> @peterbarker Looking at a log file I have with a transition from VTOL to
> cruise, ATT.DesYaw goes to 0 after the transition to cruise. Is that
> expected? Roll and Pitch are logged just fine in both modes.

It's expected:

https://github.com/ardupilot/ardupilot/blob/master/ArduPlane/Log.cpp#L16