ArduPilot/ardupilot #5303: ArduPlane:QuadPlane: Altitude loss when switched to QHOVER from QSTABILIZE

ferdicakmak, 2016-11-30:
#### Issue details
Hello,

When I switch to QHOVER from QSTABILIZE the aircraft looses altitude around 1m then recovers to commanded altitude. I experienced this in all of the flights performed with ArduPlane V3.7.1. Logs show that QTUN.ThrOut drops down to 0.13 from 0.51 (in QSTABILIZE mode) then increases to 0.69 to gain altitude then everything becomes normal.

[image]

I also checked QTUN.DCRt and QTUN.CRt and see that desired climb rate is ok. And actual climb rate reaches negative values with altitude loss due to throttle drop.

[image]

I believe that this may be caused by hover throttle estimate which is available in ArduCopter (CTUN.ThrH). If this parameter may be added to logs of ArduPlane code for QuadPlane, it would be easier to identify the problem.

Any suggestions or improvements for addressing this problem?

Thanks,
Ferit.

#### Version
ArduPlane V3.7.1

#### Platform
[  ] All
[  ] AntennaTracker
[  ] Copter
[X] Plane
[  ] Rover

#### Airframe type
QuadPlane

#### Hardware type
Pixhawk

#### Logs
-

magicrub, 2016-11-30:
yeah, that doesn't look right. Thanks for the bug report!

Naterater, 2019-01-08:
There have been a lot of changes to the quadplane code since 2016. Is the behavior repeatable today on 3.9?

IamPete1, 2021-07-25:
We have hover throttle learning on plane now, https://github.com/ArduPilot/ardupilot/pull/10323

closed 2021-07-25 by IamPete1