ArduPilot/ardupilot #5312: Processing the PARAM_REQUEST_READ MAVLink message takes 8ms, blocks main loop

jschall, 2016-12-02:
Discussed this with Tridge. It is likely AP_Param::find taking the majority of the time.

When a GCS sends a lot of PARAM_REQUEST_READ messages, I've seen it make delay the 400hz loop for 180ms.

tridge, 2016-12-02:
This is related to an issue Peter found with slow parameter saves. I think the soln is to create an async param find interface.
The interface would be something like AP_Param::find_async(). It would take a callback structure and the find would happen in the IO thread. We'd limit the number of pending requests to a reasonable number (maybe 20 or so?). I think we'd implement it as a linked list, and the callback structure would have the list pointers builtin. We can probably make it a simple singly linked list with a semaphore.
We'd also have a bitmask of dirty parameters, and the same IO thread callback in AP_Param would write out dirty parameters. That would solve the problem Peter found.

magicrub, 2016-12-02:
That statement has a ridiculously high number of nerdy keywords.

peterbarker, 2016-12-02:
This is fun :-)

> We'd limit the number of pending requests to a reasonable number (maybe 20 or so?

And when we reach that limit we....
(a) go synchronous (probably a *really* bad idea if the IO thread isn't playing nice)
(b) call the callback with a "failed" flag set?
(c) eject and let the passengers sort it out

I'm also a touch concerned about the context in which the callback is called.  I dont think it is just the main thread which sets parameters any more, so having a periodic() call into AP_Param from the main thread might not be sufficient.

peterbarker, 2016-12-03:
@jschall Which GCS is sending a whole bunch of `PARAM_REQUEST_READ`s?

I've got some patches here that implement things kinda like what tridge described for `PARAM_REQUEST_READ`.

MAVProxy doesn't tend to `PARAM_REQUEST_READ`; it uses `PARAM_REQUEST_LIST` to get them in an efficient stream.  QGC appears to act similarly...

WickedShell, 2016-12-03:
All GCS's have to recover from any lost parameters on the list with the
_read. Typically that means at least 10 parameters in my experience.

DonLakeFlyer, 2016-12-03:
When going over radio or WiFi it's rare that QGC does not lose a parameter along the wire from the PARAM_REQUEST_LIST. The number of re-requests it needs to make varies quite a bit. I'd say 10 is more of a lower bound. 

There are other cases where QGC requests batches of PARAM_REQUEST_READ. This happens after various sensor calibrations steps where it wants to get back to calibration values. For example after compass cal, QGC asks to get back all COMPASS_* values. When it does this it has an internal batching mechanism to prevent flooding the pipe. It batches 10 requests at a time until it has all of them.

WickedShell, 2016-12-03:
Also required after working with rally points if you change the total
number of rally points, before moving on to actually setting rally points
(which playing with this in flight is not all that rare/edge case)

OXINARF, 2017-06-02:
I think this was fixed with the recent work done by @tridge and @peterbarker. If you feel otherwise feel free to re-open.

closed 2017-06-02 by OXINARF