ArduPilot/ardupilot #5313: better handling of receiver failsafe

tridge, 2016-12-02:
When using a RX that has failsafe reporting (such as SBUS or SRXL) we currently consider a receiver in failsafe to be the same as a dead receiver. This means you can't do "receiver failsafe" setups for planes, where you pre-program particular settings (such as RTL) in the receiver.
I think if THR_FAILSAFE=0 then we should still take receiver inputs for receivers in failsafe. We can change hal.rcin to return an enum instead of boolean for receiver state.

R-Lefebvre, 2016-12-04:
This is a good point Tridge.  I've started using a Futaba system recently, and I've wondered about this.  It's an older FAAST system. The radio Tx only gives me the option of setting failsafe channel outputs, it doesn't allow for simply no-signals.  So I had to set it up for traditional throttle failsafe.  I know this is less robust than no-pwm, so I would be interested to use a system like you propose.

kd0aij, 2017-03-15:
I just ran into a problem with and FrSky XSR sbus-only RX on a PixRacer: for some reason this one will only output 1500 on all channels with the TX off, regardless of the TX values present when you push the F/S button on the RX. Could be a bad button, or just a different version of firmware in the XSR.

I tried 2 workarounds in the FMU code: stop publishing ORB messages or set raw_rc_values[0:3] to zero  when the sbus failsafe flag is set: both resulted in immediate disarm with FS_THR_ENABLE=3.
I was expecting a transition to LAND mode...

Noircogi, 2018-08-23:
If the receiver is configured to send RSSI over S.Bus, or the user is planning to drive channels to specific values during failsafe, it will work over PPM, but not S.Bus due to this issue.
For example, with the system running and RSSI sent via S.Bus, the OSD will continue to show the last "good" RSSI value if the transmitter is turned off.

tridge, 2019-03-15:
fixed with PR #10784

closed 2019-03-15 by tridge