ArduPilot/ardupilot #5314: GPS_INPUT not working with receiving message from mavros_extras plugin

291, 2016-12-02:
#### Issue details
Now, I'm trying to make new plugin for mavros_extras. I want to make fake gps plugin like a "mocap_fake_gps.cpp". I try to send message "GPS_INPUT message".

This is my code to send GPS_INPUT. (I edited mocap_fake_gps.cpp line.135-149 )

```
        gps_week_ms = 401400000 + (count * 200); //gps_week_sec 401400 (2016/12/1 15:30:00)

        mavlink::common::msg::GPS_INPUT pos;
        pos.time_usec = trans->header.stamp.toNSec() / 1000;        
        pos.gps_id = 2; //ID of the GPS for multiple GPS inputs
        pos.ignore_flags = 1;
        pos.time_week_ms = gps_week_ms;
        pos.time_week = 1925;   //2016-12-01    GPS week number
        pos.fix_type = 3;
        pos.lat = (lat_rad * 180 / M_PI) * 10000000;
        pos.lon = (lon_rad * 180 / M_PI) * 10000000;
        pos.alt = (408 - down) * 1000;
        pos.hdop = 0.02;
        pos.vdop = 0.02;
        pos.vn = vn;
        pos.ve = ve;
        pos.vd = vd;
        pos.speed_accuracy = 1;
        pos.horiz_accuracy = 5;
        pos.vert_accuracy = 3;        
        pos.satellites_visible = 6;
        UAS_FCU(m_uas)->send_message_ignore_drop(pos);
```

And I publish rostopic "/mavros/fake_gps/fix".
`rostopic pub -r 5 /mavros/fake_gps/fix geometry_msgs/TransformStamped '{header: auto, child_frame_id: /body, transform: {translation: {x: 0, y: 0, z: 0}, rotation: {x: 0, y: 0, z: 0, w: 1}}}'`

I set parameter.
`GPS_TYPE,14
SERIAL2_BAUD,57
SERIAL2_PROTOCOL,1`

but I can't find this parameter on Erle Brain 2.
`BRD_SER2_RTSCTS,0`

I can't see the GPS_INPUT message (using APM Planner 2)
[image]

What's wrong my code? or APM can't use GPS_INPUT ?

#### Version
APM:Copter V3.4-rc2 (f46a31d0)

#### Platform
[  ] All
[  ] AntennaTracker
[ x ] Copter
[  ] Plane
[  ] Rover

#### Airframe type
Quad

#### Hardware type

Elle Brain 2 (ErleCopter)
[http://docs.erlerobotics.com/brains/erle-brain-2](url)

291, 2016-12-02:
Sorry, I can't know where I post this issue.
So, I posted same issue on ArduPilot forum(http://discuss.ardupilot.org/).
If I wrong where to post, please reply me on ArduPilot forum.
[http://discuss.ardupilot.org/t/gps-input-not-working-with-receiving-message-from-mavros-extras-plugin/13256](url)

OXINARF, 2016-12-03:
The forum or Gitter (http://gitter.im/ArduPilot/ardupilot) is the right place for this question.

I don't use APM Planner but it looks like that screen is showing the MAVLink messages received from ArduPilot. You won't see a GPS_INPUT message coming from ArduPilot, that message has to be directed to ArduPilot. If you want to see if it is working you have to look for other MAVLink messages with the position.

closed 2016-12-03 by OXINARF