ArduPilot/ardupilot #5318: FrSKY: error msg needed when multiple ports configured

magicrub, 2016-12-02:
FB discussion: https://www.facebook.com/groups/ArduPilot.org/permalink/1269705836385940/

From @tomlauzon
> You can only have one [SERIAL_n_PROTOCOL] with FrSky at a time. Set the other ones to something other than 3 or 4. 

If this is not supported, or causes problems, then it should give some sort of GCS error and possibly inhibit arming so the user knows how to self-correct it and does not fly the aircraft in an unstable or unsupported configuration.

floaledm, 2016-12-02:
Thanks Tom. The way it's setup now (and that's how the lib was originally written), the first serial port configured for FrSky (currently either 3, 4 or 10) will be active and the other serial ports configured for FrSky will essentially not do anything.

My understanding of the code is that AP_Mount, AP_RangeFinder, AP_Proximity and AP_Beacon all have backends to support multiple instances.

Seems to me like this approach may be preferable... thoughts?

magicrub, 2016-12-02:
Only supporting a single one is fine, unless you can think of a use-case to support multiple. I can't. The problem is if serial2,3,4,5 are configured and it's hooked up to 4/5 then we get situations like what we see on that FB thread. I think a simple GCS text message saying:
 "FrSky serial 3 init skipped, already running on 2".
 "FrSky serial 4 init skipped, already running on 2".
 "FrSky serial 5 init skipped, already running on 2".

floaledm, 2016-12-03:
Yes, there's not really a situation in which we'd need more than one FrSky serial. I like your suggestion as it's pretty straightforward. I could loop on using serial_manager.find_serial() to look for the second, third,... instances for each of the FrSky protocols (e.g., `serial_manager.find_serial(AP_SerialManager::SerialProtocol_FrSky_D, 1)`. Am I on the right track?

Maybe you can clear something out for me here. Looking at how the serial port is initialized in other libs, from what I can tell, they always look for the first instance (`serial_manager.find_serial(AP_SerialManager::[serial_protocol_name], 0)`, except for the GPS, where it looks for the first two instances `serial_manager.find_serial(AP_SerialManager::SerialProtocol_GPS, 0)` and `serial_manager.find_serial(AP_SerialManager::SerialProtocol_GPS, 1)`.

Do you know if one can actually have more than one of each of a gimbal, rangefinder, lidar360, or beacon on serial ports (e.g., two serial rangefinders - SerialProtocol_Lidar)?

magicrub, 2016-12-03:
I don't know, never looked into it. Perhaps you can write a good example that we can add to all serial based protocols

closed 2024-08-05 by tridge