ArduPilot/ardupilot #5323: AP_Landing: Lift all the landing flight stages into just stage LAND

WickedShell, 2016-12-04:
#### Issue details
AP_Landing should manage all the finer details of the different steps in a landing, and there should only be a single FLIGHT_STAGE_LAND which signifies to call into AP_Landing to do work. This is needed to allow other landing types.

#### Platform
[  ] All
[  ] AntennaTracker
[  ] Copter
[ x ] Plane
[  ] Rover

magicrub, 2016-12-04:
I like that. Each landing type has its own state machine and mini stages. What about flaps? Right now they're purely airspeed controlled, think that should go down in there too?

WickedShell, 2016-12-04:
I think it should. float AP_Landing::get_flap_percentage() comes to mind...
(I think its a percent that it wants right?) Would allow the different
landing types to pick what they need easily, and possibly apply it
elsewhere (not sure why they would want to apply it elsewhere yet, but need
to be able to opt out of it if you don't want them)

WickedShell, 2017-02-28:
This was done, we never closed the issue.

closed 2017-02-28 by WickedShell