ArduPilot/ardupilot #5326: Plane: QuadPlane sluggish at WP before transition

expntly, 2016-12-04:
#### Issue details
On a mission with a first VTOL part where the vehicle is asked to move to a WP and then transition to fixed-wing, it will stay at that WP for a while before it transitions.  Sometimes it stays stuck there for up to 20s.

Here's a mission example.

[image]

The expected behavior is that it starts transitioning as soon as it enters the WP radius (or passes the orthogonal line) without any delay.

#### Version
3.7.0

#### Platform
[  ] All
[  ] AntennaTracker
[  ] Copter
[X] Plane
[  ] Rover

#### Airframe type
QuadPlane

#### Hardware type
Pixhawk

#### Logs

It can be reproduced by creating a mission such as the one above.

expntly, 2016-12-06:
@OXINARF I wouldn't categorize it as an "Enhancement".  There's an issue somewhere, I haven't figured what it is yet.

OXINARF, 2016-12-06:
@expntly Not saying it isn't an issue, but it looks like the system works although it can be improved/enhanced to remove the delay.

IamPete1, 2021-08-15:
I cant replicate this in SITL, I'm going to close, if someone has a log that shows the issue on 4.1 we should re-open.

closed 2021-08-15 by IamPete1