ArduPilot/ardupilot #5328: Hal_Px4: SPIDevice : Wrong placement for nullptr

khancyr, 2016-12-05:
#### Issue details
Looking at https://github.com/ArduPilot/ardupilot/blob/master/libraries/AP_HAL_PX4/SPIDevice.cpp#L257
The check of device_table[i].name == nullptr should be before use strcmp(device_table[i].name, name right ?

OXINARF, 2016-12-05:
What value for *i* would you use? :stuck_out_tongue_winking_eye:
Although not the best coding pattern, the table has a null entry in the end. What this does is go through the entire table and search for the name. If after the search the reached value is a null name then we know we haven't found it.

closed 2016-12-05 by OXINARF

khancyr, 2016-12-06:
@OXINARF ok ! See my error!  if device_table[i].name == nullptr then it exit the for loop then it is ok !