ArduPilot/ardupilot #5335: Plane: QuadPlane transition to VTOL causes pitch up and significant alt increase

expntly, 2016-12-06:
#### Issue details
When transitioning from fixed-wing to VTOL the vehicle is pretty nervous.  What happens is the vehicle pitches up rather aggressively as the quad motors are set in motion.  Alt gain is around +10m.  Setting the last WP before VTOL_LAND farther away (140m) doesn't seem to help much.

Graph from SITL simulation (see log below):
[image]

#### Version
3.7.0

#### Platform
[  ] All
[  ] AntennaTracker
[  ] Copter
[X] Plane
[  ] Rover

#### Airframe type
QuadPlane

#### Hardware type
Pixhawk

#### Logs

[sitl23.BIN.zip](https://github.com/ArduPilot/ardupilot/files/632911/sitl23.BIN.zip)

It's easy to reproduce in SITL with an AUTO mission: 1) VTOL_TAKEOFF, 2) WAYPOINT (70m before next), 3) VTOL_LAND.

expntly, 2017-09-17:
I just tested this on 3.8.2-dev SITL, this is still happening.  Is there a way to mitigate this behavior?  This causes quite some stress on the airframe.  Ideally the VTOL motors would rev up slowly as the plane slows down.

magicrub, 2017-09-20:
still on the TODO list, sorry it's taking so long to look into.

expntly, 2017-10-04:
Is there a parameter similar to THR_SLEWRATE but for the quad motors?

expntly, 2017-12-07:
Any update on this?  We're see high g loads in Z acceleration at de-transition.

IamPete1, 2021-08-15:
This is use should be much improved with the updated transition logic. https://github.com/ArduPilot/ardupilot/pull/14938

I'm going to close for now, if this issue is still present on 4.1 please reopen.

closed 2021-08-15 by IamPete1